TheThingsNetwork/lorawan-stack #3406: Sending gateway PUT request with empty diff results in an error

pgalic96, 2020-10-26:
#### Summary

When being in entity general settings, we allow submitting PUT requests with no changes for applications. Currently submitting the same PUT request with empty payload for gateways results in a backend error:

#### Steps to Reproduce

1. Submit a request with to `http://localhost:1885/api/v3/gateways/<gateway_id>` with request payload `{"gateway":{},"field_mask":{"paths":[]}}`
2. Observe the error returned

#### What do you see now?

```
{"code":3,"message":"error:pkg/errors:validation (invalid `gateway`: embedded message failed validation)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/errors","name":"validation","message_format":"invalid `{field}`: {reason}","attributes":{"field":"gateway","name":"UpdateGatewayRequestValidationError","reason":"embedded message failed validation"},"correlation_id":"27e3f23d15f946f5a0e32d3988310944","cause":{"namespace":"pkg/errors","name":"validation","message_format":"invalid `{field}`: {reason}","attributes":{"field":"version_ids","name":"GatewayValidationError","reason":"embedded message failed validation"},"cause":{"namespace":"pkg/errors","name":"validation","message_format":"invalid `{field}`: {reason}","attributes":{"field":"model_id","name":"GatewayVersionIdentifiersValidationError","reason":"value does not match regex pattern \"^[a-z0-9](?:[-]?[a-z0-9]){2,}$\""},"code":3},"code":3},"code":3}]}
```

#### What do you want to see instead?

Success like with submitting exactly the same payload for application PUT request

#### Environment

- `v3.10.0-rc1` 

#### How do you propose to implement this?

Don't know exactly

#### How do you propose to test this?

Testing in console

#### Can you do this yourself and submit a Pull Request?

I think that would take too much time

htdvisser, 2020-10-27:
This is actually working according to the [specification of a field mask](https://developers.google.com/protocol-buffers/docs/reference/google.protobuf#field-masks-in-update-operations). If an update request doesn't specify the fields to update, the operation applies to all fields. In this case, validation fails for one of the (implied) fields to update.

pgalic96, 2020-10-27:
Then I guess the application validation does not work as intended?

htdvisser, 2020-10-27:
I guess you mean gateway.

If you update all fields of a gateway, you also update the "model ID", which must follow a specific pattern, and apparently an empty string is not valid.

pgalic96, 2020-10-27:
> When being in entity general settings, we allow submitting PUT requests with no changes for applications. Currently submitting the same PUT request with empty payload for gateways results in a backend error:

The following PUT request for applications succeeds, with preserving all fields:
`http://localhost:1885/api/v3/applications/my-new-app`

Payload: `{"application":{},"field_mask":{"paths":[]}}`

The response is following:

`{"ids":{"application_id":"my-new-app"},"created_at":"2020-07-06T13:55:28.598Z","updated_at":"2020-10-27T11:30:14.240Z"}`

I expected the same behavior for the gateway PUT requests, but as you said, the validation fails for one of the fields. Is it working as intended for both gateways and applications or is the validation not consistent?

htdvisser, 2020-10-27:
I suspect that for the application, all fields are allowed to be empty.

Maybe we should actually reject update requests with empty field masks. In any case the Console should avoid making requests with empty field masks.

htdvisser, 2020-10-27:
Adding @johanstokking and @rvolosatovs (as code owners of the API) for discussion.

Should we update our validation to **require** a field mask to be set on update requests?

The problem is that it technically breaks API, but I think it's justified as a bugfix, since the current behavior is confusing, and could lead to unexpected results when we add fields in newer versions, while a client remains at an old version, and may unintentionally zero out a field by supplying it in the "body" with an empty field mask.

pgalic96, 2020-10-27:
The change will break application general settings, so that needs to be taken into account.

pgalic96, 2020-10-27:
So, this is handled differently for pretty much every entity general settings form in the Console. @bafonins pointed it out in #3364 , but actually submitting a form with no fields changed in the organization general form also results in an error (albeit a different one because of a different field mask `[""]` ):

[image]

The way it is handled in the device general settings is correct imo, where JS SDK handles this with a request tree. If there is no updated fields, no requests are sent at all. So we either change the SDK to handle this in the same way for all the entities, or another solution would be to disable the submit button if no fields are changed, which sounds like a way simpler fix, that is also satisfying UX requirements. cc @kschiffer

johanstokking, 2020-10-28:
> @htdvisser  and could lead to unexpected results when we add fields in newer versions, while a client remains at an old version, and may unintentionally zero out a field by supplying it in the "body" with an empty field mask.

If the field mask is empty, the value should not be taken from the body, right?

I think that supplying an empty field mask should be a no-op on update. It could be used by clients to ensure the entity exists and/or that there's valid auth to make an actual change.

The issue reported by @pgalic96 seems to be that we require certain fields on `PUT`, but I don't think that we should be doing that.

htdvisser, 2020-10-28:
> If the field mask is empty, the value should not be taken from the body, right?
> [...]
> I think that supplying an empty field mask should be a no-op on update

This would violate the protobuf specification of a field mask which says that if an update request doesn't specify the fields to update, the operation applies to all fields.

Our validation middleware respects the protobuf specification, and therefore validates the fields specified in the field mask, or all fields if the field mask is empty.

johanstokking, 2020-10-28:
But you are suggesting to deviate from protobuf specification, right, by not allowing empty field masks?

There may be other issues when not supporting empty field mask, especially on create, i.e. when there are no fields to be set.

htdvisser, 2020-10-29:
Requiring a field mask is allowed by the specification:

> Note that in the presence of schema evolution, this may mean that fields the client does not know and has therefore not filled into the request will be reset to their default. If this is unwanted behavior, a specific service may require a client to always specify a field mask, producing an error if not.

nejraselimovic, 2021-04-19:
@htdvisser @johanstokking what is the conclusion here, anything left for discussion? Who is picking up this issue at the end? Can we move it from Backlog?

johanstokking, 2021-04-20:
@pgalic96 is this still an issue?

pgalic96, 2021-04-20:
It does still occur in organization general settings if you submit without changing anything. For gateway general settings the attributes field is always included in the field mask even when not changed so there is no error but I suppose the issue is not fixed exactly.

johanstokking, 2021-04-21:
> another solution would be to disable the submit button if no fields are changed, which sounds like a way simpler fix, that is also satisfying UX requirements

This seems like the best approach then.

closed 2021-04-21 by johanstokking