TheThingsNetwork/lorawan-stack #3410: Document that location is only updated automatically for authenticated gateways

neoaggelos, 2020-10-27:
#### Summary

Followup for #3326 

#### Why do we need this ?

Gateway location can only be updated automatically from status messages when the gateway is authenticated, which excludes gateways connecting over UDP.

#### What is already there? What do you see now?

Comment on the API proto field `update_location_from_status`, noting that it does not work for UDP gateways

#### What is missing? What do you want to see?

Document this behaviour in the docs and the console (also see comment https://github.com/TheThingsNetwork/lorawan-stack/pull/3326#discussion_r511057626)

#### Can you do this yourself and submit a Pull Request?

No, assigning @benolayinka, re-assign as needed.

benolayinka, 2020-11-17:
@kschiffer it would be nice to add a note in the console for this

benolayinka, 2020-11-17:
Proto comments go in the stack definition. Moving this issue.

benolayinka, 2020-11-17:
@htdvisser can you add the proto comment

htdvisser, 2020-11-17:
No, you can do this yourself.

benolayinka, 2020-11-17:
Comment is already there https://github.com/TheThingsNetwork/lorawan-stack/blob/cb3838f1778850bc069ac8486297db7683836ac3/api/gateway.proto#L133.

This is probably more important for people to see in the console, assigning @kschiffer

kschiffer, 2021-06-01:
@bafonins can you add this to the respective tooltip in #4157

kschiffer, 2021-06-01:
Will be fixed by #4157. Closing this already.

closed 2021-06-01 by kschiffer