TheThingsNetwork/lorawan-stack #3413: CLI support for Join Server Application Activation Setting Registry

rvolosatovs, 2020-10-27:
#### Summary

Console support for https://github.com/TheThingsNetwork/lorawan-stack/pull/3277. 

#### Why do we need this?

To actually use it without direct API calls.

#### What is already there? What do you see now?

Only API implementation

#### What is missing? What do you want to see?

Console support

#### How do you propose to implement this?

Add a tab in application settings

#### How do you propose to test this?

Integration testing

#### Can you do this yourself and submit a Pull Request?

Can review

kschiffer, 2021-08-16:
cc @johanstokking we are lacking some details on this. Can you help us out seeing whether this is still necessary?

johanstokking, 2021-08-17:
That's what you get when one doesn't take some time to write issues 🤷‍♂️ 

This one is about Console support for the application activation settings (AAS). They are used by the Join Server when activating a device. It's about the following settings:

1. Key encryption key and label. This is used to wrap the AppSKey before it is sent to the Network Server. This way, the Network Server cannot see the the AppSKey. The Application Server has the same KEK configured in its key vault. The label is used to identify the key (and load it from the key vault)
2. Home NetID. This overrides the NetID if configured for the end device.
3. Application Server ID. This is only used when using LoRaWAN backend interfaces, to authenticate the Application Server

See https://github.com/TheThingsNetwork/lorawan-stack/blob/4d9bf10ba5ed8056940817f411fe1cd21ec06d2e/api/joinserver.proto#L241-L290 for the API.

johanstokking, 2023-02-23:
This is no longer necessary with The Things Join Server. We may consider removing AAS from the in-cluster Join Server. I'll file a follow up issue.

closed 2023-02-23 by johanstokking