TheThingsNetwork/lorawan-stack #3426: CUPS - Server is not forwarding credentials and trust CA to the gateway

ogimenez-smtc, 2020-10-30:
#### Summary
CUPS server is not providing credentials (`Authorization: Bearer XXXX`) to the gateway
CUPS server is not providing trust certificate to the gateway

#### Why do we need this?
Following documentation it seems that TTI recommend to use HTTP header `Authorization` over TLS credentials (covered by #137), but I am not able to make it work: CUPS server is only providing the websocket URI

#### Environment

TTI hosted lorawan-stack, v3.9.4

#### How do you propose to implement this?
lorawan-stack should provide trusted CA
lorawan-stack should provide `Authorization` token if API key used for CUPS call allows to create a new API key

#### How do you propose to test this?
Provide only CUPS files to Basic Station and start it. It should fetch required LNS data by itself

#### Can you do this yourself and submit a Pull Request?
No

KrishnaIyer, 2020-10-30:
With #2865, this is quite easily doable. I'll add some documentation.

KrishnaIyer, 2020-12-16:
This is actually closed with https://github.com/TheThingsNetwork/lorawan-stack/pull/2865.

Please refer to the updated documentation here; https://www.thethingsindustries.com/docs/gateways/lora-basics-station/cups/

closed 2020-12-16 by KrishnaIyer

Oliv4945, 2020-12-17:
Thank you @KrishnaIyer , I'll test it.

For your information the following warning is wrong:

```
Configuring CUPS automatically retrieves LNS credentials and configures LNS on your gateway. If you configure CUPS, DO NOT manually configure LNS, or your gateway may ignore the CUPS credentials. 
```
I propose:
```
Configuring CUPS automatically retrieves LNS credentials and configures LNS on your gateway. If you configure CUPS, your gateway will override LNS credentials in case of successful CUPS connection. 
```

KrishnaIyer, 2020-12-17:
Sounds good; @benolayinka