TheThingsNetwork/lorawan-stack #3433: Integrate Device Repository

johanstokking, 2020-11-03:
#### Summary

Integrate the LoRaWAN Device Repository: https://github.com/TheThingsNetwork/lorawan-devices.

This issue spans onboarding new end devices.

References #2813 

@bafonins is the implementer.

@kschiffer please design and implement Console UX for browsing end devices by brand, model and potentially firmware and hardware versions. Note that there will be thousands of end devices.

#### Why do we need this?

To make onboarding LoRaWAN devices easy.

#### What is already there? What do you see now?

Device creation wizard and general settings page for advanced device settings. This is quite overwhelming and error prone to users who have a "LoRaWAN device" in their hands and they only know the vendor and model.

#### What is missing? What do you want to see?

The user journey would be as follows:

1. User clicks **Add end device**
2. User gets to browse devices by brand, model, firmware version (if there are multiple) and hardware version (if there are multiple [supported hardware versions in the firmware version](https://github.com/TheThingsNetwork/lorawan-devices/blob/782727a8a99c150fda616b1293980b20e6796ee6/vendor/example/windsensor.yaml#L16-L18)). Below, the user can still start the current wizard for manual creation
3. When the user selected their brand, model and version, they still have to enter the device identifiers (for OTAA: JoinEUI, DevEUI and root key(s); for ABP: DevEUI, DevAddr and session keys). Whether the end device is OTAA or ABP is in the device repository. Also the MAC version is in the device repository, so only the relevant root key(s) and session keys are entered by the user
4. Create the end device from the template, also setting the payload formatter to repository

#### How do you propose to implement this?

1. Remove `EndDeviceVersion` message from the API; this is unused and superseded by `EndDeviceTemplate`. Consider also removing `EndDeviceBrand` and `EndDeviceModel`, unless you need it for the Device Repository API (see point 3 below)
2. Create a new first-class component Device Repository
3. Implement API, config and a service implementation for browsing end devices from the repository. This is much like the existing Webhook Templates (which are also being loaded as YAML), although we probably need **search**, **pagination**, loading a profile and codecs by `EndDeviceVersionIdentifiers` and serving static assets (like photos). As reference:
   1. API: https://github.com/TheThingsNetwork/lorawan-stack/blob/6f15ec162081c282a370008af834811d0df76efa/api/applicationserver_web.proto
   2. Config: https://github.com/TheThingsNetwork/lorawan-stack/blob/6f15ec162081c282a370008af834811d0df76efa/pkg/applicationserver/io/web/config.go
   3. Store: https://github.com/TheThingsNetwork/lorawan-stack/blob/6f15ec162081c282a370008af834811d0df76efa/pkg/applicationserver/io/web/templates.go
   4. Service implementation: https://github.com/TheThingsNetwork/lorawan-stack/blob/6f15ec162081c282a370008af834811d0df76efa/pkg/applicationserver/io/web/grpc_webhooks.go#L63-L69
   5. Webhook templates do not support static assets at the moment, but this would be similar to https://github.com/TheThingsNetwork/lorawan-stack/issues/3357
4. The API in principle doesn't require rights, but do require any authentication
5. The API produces an `EndDeviceTemplate` (like when importing devices). The field mask covers the LoRaWAN device profile for either OTAA or ABP [1]. Also, the [`version_ids`](https://github.com/TheThingsNetwork/lorawan-stack/blob/6f15ec162081c282a370008af834811d0df76efa/api/end_device.proto#L409-L410) should be set. Do **NOT** copy the payload codec contents in each end device: instead, set `formatters.{up,down}_formatter` to `FORMATTER_REPOSITORY`, and Application Server will load it from the Device Repository
6. Implement a client in the JS SDK
7. Browse the device repository in the Console

[1] Here's an example device profile for OTAA:

```json
{
  "supportsClassB": false,
  "supportsClassC": false,
  "macVersion": "1.0.2",
  "regionalParametersVersion": "RP001-1.0.2-RevB",
  "supportsJoin": true,
  "maxEIRP": 14,
  "supports32bitFCnt": true
}
```

And here's one for ABP:

```json
{
  "supportsClassB": false,
  "supportsClassC": false,
  "macVersion": "1.0.4",
  "regionalParametersVersion": "RP002-1.0.1",
  "supportsJoin": false,
  "rx1Delay": 1,
  "rx1DataRateOffset": 0,
  "rx2DataRateIndex": 3,
  "rx2Frequency": 869.525,
  "factoryPresetFrequencies": [
    868.1,
    868.3,
    868.5,
    867.1,
    867.3,
    867.5,
    867.7,
    867.9
  ],
  "maxEIRP": 14,
  "supports32bitFCnt": true
}
```

These fields map 1-to-1 to fields in the `EndDeviceTemplate`. @rvolosatovs can assist further on this.

#### How do you propose to test this?

Unit testing for the store, service implementation etc, like webhook templates.

Integration testing for everything else.

#### Can you do this yourself and submit a Pull Request?

Can review and test, can also answer questions if any.

rvolosatovs, 2020-11-03:
@johanstokking so this will be handled fully by NS clients then and NS itself should not use device repository itself? I.e. https://github.com/TheThingsNetwork/lorawan-stack/blob/ceb68e8d1a0e32dbbb222eb82a42ceed2b3e1169/pkg/networkserver/grpc_deviceregistry.go#L373-L375 this is not going to be used, is that correct?

johanstokking, 2020-11-03:
Yes, good point, that is correct. The clients will work with the DR component directly and create the `EndDevice` in the respective device registries as normal. This works like importing devices works too.

benolayinka, 2020-11-03:
Walked through adding a device and have the following comments:

- Schema doesn't check for missing `firmwareVersions` though it's required
- Schema validation errors could use a one-liner printing the error as human readable, i.e "Schema error in index.yml. Value 'example.com' of field 'website' is invalid."

Currently looks like:

```
./vendor/index.yaml is invalid
[
  {
    keyword: 'format',
    dataPath: '.vendors[0].website',
    schemaPath: '#/properties/website/format',
    params: { format: 'uri' },
    message: 'should match format "uri"'
  }
]
```

johanstokking, 2020-11-03:
Fix that, thanks. Output is now:

```
$ make validate
npm run validate

> lorawan-devices@1.0.0 validate
> node bin/validate.js

./vendor/index.yaml is invalid: .vendors[28].website should match format "uri"
```

johanstokking, 2020-11-12:
Looped in @htdvisser for generating an index.

neoaggelos, 2020-11-13:
>  This is much like the existing Webhook Templates (which are also being loaded as YAML)

After glancing at the webhooks templates registry implementation, I understand that the webhook templates are cached and fetched once every X minutes. Is this acceptable for the device repository as well? Could this pose scalability issues if we plan to support searching and indexing for thousands of devices?

In any case, we can always start simple and improve on it with future iterations.

johanstokking, 2020-11-13:
@htdvisser therefore suggested to generate an index file as part of GitHub actions. That index file can be loaded in Bleve Search.

The DR component would fetch a new index file in the background. Ideally, it would use a URL with cache headers, so it can fetch it frequently.

htdvisser, 2020-11-17:
Yes, so my idea was to use [bleve](https://github.com/blevesearch/bleve) for searching. The search index would be built (see [indexing](https://github.com/blevesearch/bleve#indexing)) by CI of the device repository and uploaded somewhere. The Things Stack periodically checks for an updated search index (this definitely doesn't have to be every X minutes, more like daily), downloads it and (re)loads it for querying (see [querying](https://github.com/blevesearch/bleve#querying)).

neoaggelos, 2020-11-18:
A few notes after doing some initial digging:

Indexing:
- bleve does not work with incomplete words. (e.g. searching for `thing` will not match `The Things Network`)
- bleve works well for searching/sorting/pagination, but only returns ids, so we need a second step that will fetch the actual vendor/device information.
- in that sense, I think we should have all the bleve-related stuff in TTS. e.g. let's add a `ttn-lw-stack device-repo create-index` command, and use that with the device repo CI.
- a bleve index is a folder (not a file), and it creates temporary files (e.g. locks), so this needs extra logic on top of the fetcher interface. A simple way to overcome this would be to fetch a zipped archive, unzip locally and use that, but that is not too clean. What other options do we have have? Note that the index file itself may be very large.

Protos:
- The device repository is using the term "vendor", however the existing `EndDeviceVersionIdentifiers` is using "brand". Switching to `vendor` is a breaking change (both in API and in the DB schema), but using a mixture of brand and vendor can cause confusion to users. How should we proceed with this? For now, I am using `brand` for all user-facing messages.
- Started with removing `EndDeviceVersion` protos, and added a new `string Region` field to the `EndDeviceVersionIdentifiers` proto.
- Will use protocol buffers for all user-facing APIs, most of them following the existing schema from the Device Repository. Internally, we do not use the protocol buffers for loading, but instead use structs with `yaml` tags for fields that we care about.
- Should any user-facing fields use enums instead of string fields? (e.g. for device sensor types).
- [Here is some initial work on the devicerepository protos](https://github.com/TheThingsNetwork/lorawan-stack/commit/449de96cf44f2a440931e47d407b84ea1fda9f07), but some stuff is still TODO.

API:
- The design is driven by the initial console needs. It allows listing vendors, listing end device definitions for a specific vendor, and retrieving the end device template using vendor_id, model_id, firmware_version and region as inputs. Also an API for retrieving the message payload formatters, which probably does not need an HTTP API.
- We can improve on this, and even support stuff that might be needed for the wall of fame as well, so that we re-use functionality as much as possible.

Implementation:
- The DeviceRepository component uses a `Store` interface to work with the device repo. The naive store simply uses a `fetch.Interface` to retrieve files from the device repository, and we can have an `indexedStore` on top which uses bleve for searching/sorting/pagination, and then only fetches the required information from the device repo.

I have started working on the implementation based on the above, please comment with any objections, suggestions, ideas. I think we should decide on the API early enough to avoid wasting development and review time with iterations on the pull request.

htdvisser, 2020-11-18:
> bleve does not work with incomplete words

Can't you accomplish that with [`FuzzyQuery`](https://pkg.go.dev/github.com/blevesearch/bleve/search/query#FuzzyQuery) or something?

> but only returns ids, so we need a second step that will fetch the actual vendor/device information.

The [docs for `SearchRequest`](https://pkg.go.dev/github.com/blevesearch/bleve#SearchRequest) say _Fields describes a list of field values which should be retrieved for result documents, provided they were stored while indexing_ so this may not be necessary.

> A simple way to overcome this would be to fetch a zipped archive, unzip locally and use that, but that is not too clean. What other options do we have have?

The [Go 1.16 File System Draft Design](https://go.googlesource.com/proposal/+/master/design/draft-iofs.md) mentions mounting ZIP files as filesystems. That could be interesting.

johanstokking, 2020-11-19:
> * a bleve index is a folder (not a file), and it creates temporary files (e.g. locks), so this needs extra logic on top of the fetcher interface. A simple way to overcome this would be to fetch a [...]

> * The DeviceRepository component uses a `Store` interface to work with the device repo. The naive store simply uses a `fetch.Interface` to retrieve files from the device repository, and we can have [...]

I think it's best to use the fetcher interface only to retrieve the zipped index. I think we should work with a temporary folder to unzip it, indeed until we have a better solution with Go 1.16.

> * The device repository is using the term "vendor", however the existing `EndDeviceVersionIdentifiers` is using "brand". Switching to `vendor` is a breaking change (both in API and in the DB [...]

Indeed, use brand in TTS.

> * Started with removing `EndDeviceVersion` protos, and added a new `string Region` field to the `EndDeviceVersionIdentifiers` proto.

That should be BandID instead, in TTS terms. These should be mapped with those defined in `pkg/band`.

> * Will use protocol buffers for all user-facing APIs, most of them following the existing schema from the Device Repository. Internally, we do not use the protocol buffers for loading, but instead use structs with `yaml` tags for fields that we care about.

ACK

> * Should any user-facing fields use enums instead of string fields? (e.g. for device sensor types).

I think it's safest to go with strings

> * The design is driven by the initial console needs. It allows listing vendors, listing end device definitions for a specific vendor, and retrieving the end device template using vendor_id, model_id, firmware_version and ~region~ band as inputs. Also an API for retrieving the message payload formatters, which probably does not need an HTTP API.

ACK, but what do you mean with "which probably does not need dan HTTP API"? I think it's very useful to be able to _see_ the codec in the Console.

> * We can improve on this, and even support stuff that might be needed for the wall of fame as well, so that we re-use functionality as much as possible.

Not in TTS. Wall of Fame should not be using TTS.

neoaggelos, 2020-11-23:
Some further notes during initial implementation:

#### API

- [Here is the API](https://github.com/TheThingsNetwork/lorawan-stack/blob/feature/device-repo-api/api/devicerepository.proto) after considering the complete list of fields from the Device Repo. New messages are `EndDeviceBrand` and `EndDeviceModel`. Format closely mirrors https://lorawan-schema.org/draft/devices/1/.
- There are only `ListBrands` and `ListModels` RPCs for now, and the request uses field mask paths to retrieve the fields that we care about. There is no `GetBrand` or `GetModel`, should they be added as well?
- Currently using simple strings for `BrandID` and `ModelID` fields. Should we change this to something more elaborate?
- Regarding correctness of device attributes/MAC settings/payload formatters, the current implementation follows a best-effort approach. I believe any complex validation logic belongs in the Device Repository itself (like the jsonschema validation rules in https://lorawan-schema.org/draft/devices/1/), and not in the TTS protos.
- Do we need to list models from multiple vendors as well?
- How do we handle fields like `MaxDutyCycle`, where the Device Repo and the protos use a vastly different format?

- Should `hardware_version` be a required input for the `GetTemplate()` RPC? Strictly speaking, it is enough to specify the firmware version and the profile region. If `hardware_version` was an input, then we could simply verify that it is in the list of the hardware versions supported by the firmware. 

#### Regarding Payload Formatters

- As suggested in the description, the `GetTemplate()` RPC returns payload formatters with type `FORMATTER_REPOSITORY`, and there is a separate `GetFormatters(*ttnpb.EndDeviceVersionIdentifiers) (*ttnpb.MessagePayloadFormatters, error)` RPC for the AS (or other components) to retrieve them. Now, the device repo has 3 separate codecs (`uplinkDecoder`, `downlinkEncoder` and `downlinkDecoder`). Would it make sense to use an alternative format like so: `GetFormatters(*ttnpb.EndDeviceVersionIdentifiers, which string) (string, error)`, where `which` can be any of the codec types, and the js codec is returned? 

--- 

A couple of issues with the Device repository.

- The Things Node has invalid profile IDs (mis-match with the filenames), see https://github.com/TheThingsNetwork/lorawan-devices/pull/25
- https://github.com/TheThingsNetwork/lorawan-devices/blob/master/vendor/index.yaml#L680 says `ttn`, but The Things Node is under `the-things-products` https://github.com/TheThingsNetwork/lorawan-devices/tree/master/vendor/the-things-products. (a). Which one is correct? (b). Currently, the only way to deduce `the-things-products` as a valid vendor is to check that the directory exists. Should we do that instead of reading the `vendor/index.yaml` file?

Let me know if I should file any followup issues for these.

--- 

Example calls with the existing API:

<summary> <description>List brands</description> <details>

```js
/* GET /api/v3/dr/brands?offset=164&limit=5&field_mask.paths=vendor_id */
{
  "brands": [
    {
      "brand_id": "tektelic",
      "name": "TEKTELIC Communications Inc.",
      "vendor_id": 415
    },
    {
      "brand_id": "tta",
      "name": "Telecommunications Technology Association",
      "vendor_id": 417
    },
    {
      "brand_id": "telent",
      "name": "Telent GmbH",
      "vendor_id": 420
    },
    {
      "brand_id": "telog",
      "name": "Telog Instruments, Inc.",
      "vendor_id": 422
    },
    {
      "brand_id": "ttn",
      "name": "The Things Network Foundation",
      "vendor_id": 428
    }
  ]
}
```
</details>
</summary>

<summary><description>Get models for The Things Products vendor</description><details>

```js
/* GET /api/v3/dr/brands/the-things-products/models?field_mask.paths=name&field_mask.paths=firmware_versions */
{
  "models": [
    {
      "model_id": "the-things-node",
      "name": "The Things Node",
      "firmware_versions": [
        {
          "version": "1.0",
          "hardware_versions": [
            "1.0"
          ],
          "profiles": {
            "EU_863_870": {
              "profile_id": "the-things-node-868",
              "lorawan_certified": true,
              "codec_id": "the-things-node-codec"
            },
            "US_902_928": {
              "profile_id": "the-things-node-915",
              "lorawan_certified": true,
              "codec_id": "the-things-node-codec"
            }
          }
        }
      ]
    }
  ]
}
```

</details>
</summary>

<summary><description>Get device template for The Things Node with EU863-870 profile</description><details>

```js
/* GET /api/v3/dr/brands/the-things-products/models/the-things-node/1.0/EU_863_870/template */
{
  "end_device": {
    "ids": {
      "application_ids": {}
    },
    "created_at": "0001-01-01T00:00:00Z",
    "updated_at": "0001-01-01T00:00:00Z",
    "version_ids": {
      "brand_id": "the-things-products",
      "model_id": "the-things-node",
      "firmware_version": "1.0",
      "band_id": "EU_863_870"
    },
    "lorawan_version": "1.0.2",
    "lorawan_phy_version": "1.0.2-b",
    "supports_join": true,
    "mac_settings": {
      "supports_32_bit_f_cnt": true
    },
    "mac_state": {
      "current_parameters": {},
      "desired_parameters": {
        "max_eirp": 16
      }
    },
    "formatters": {
      "up_formatter": "FORMATTER_REPOSITORY",
      "down_formatter": "FORMATTER_REPOSITORY"
    }
  },
  "field_mask": {
    "paths": [
      "version_ids",
      "supports_join",
      "supports_class_b",
      "supports_class_c",
      "lorawan_version",
      "lorawan_phy_version",
      "formatters",
      "mac_settings.supports_32_bit_f_cnt",
      "mac_state.desired_parameters.max_eirp"
    ]
  }
}
```
</details>
</summary>

kschiffer, 2020-11-24:
> @kschiffer please design and implement Console UX for browsing end devices by brand, model and potentially firmware and hardware versions. Note that there will be thousands of end devices.

The wireframes for the device repository-powered device registration flow can be found here: https://github.com/TheThingsNetwork/lorawan-stack/issues/3522

johanstokking, 2020-11-24:
> #### API
> * [Here is the API](https://github.com/TheThingsNetwork/lorawan-stack/blob/feature/device-repo-api/api/devicerepository.proto?rgh-link-date=2020-11-23T22%3A54%3A05Z) after considering the complete list of fields from the Device Repo. New messages are `EndDeviceBrand` and `EndDeviceModel`. Format closely mirrors [lorawan-schema.org/draft/devices/1](https://lorawan-schema.org/draft/devices/1/).

Please file draft PR for API so we can review properly.

> * There are only `ListBrands` and `ListModels` RPCs for now, and the request uses field mask paths to retrieve the fields that we care about. There is no `GetBrand` or `GetModel`, should they be added as well?

Do this on an as-needed basis. Let's not over-engineer things that we don't need.

> * Currently using simple strings for `BrandID` and `ModelID` fields. Should we change this to something more elaborate?

What would be more elaborate approach be, how would what work, and why would that be necessary?

> * Regarding correctness of device attributes/MAC settings/payload formatters, the current implementation follows a best-effort approach. I believe any complex validation logic belongs in the Device Repository itself (like the jsonschema validation rules in [lorawan-schema.org/draft/devices/1](https://lorawan-schema.org/draft/devices/1/)), and not in the TTS protos.

Validation belongs in DR. Indeed, The Things Stack can assume that the DR is correct.

> * Do we need to list models from multiple vendors as well?

How do you mean?

> * How do we handle fields like `MaxDutyCycle`, where the Device Repo and the protos use a vastly different format?

These should be converted

> * Should `hardware_version` be a required input for the `GetTemplate()` RPC? Strictly speaking, it is enough to specify the firmware version and the profile region. If `hardware_version` was an [...]

Indeed, we don't need `hardware_version` as input here

> * As suggested in the description, the `GetTemplate()` RPC returns payload formatters with type `FORMATTER_REPOSITORY`, and there is a separate `GetFormatters(*ttnpb.EndDeviceVersionIdentifiers) [...]

The `FORMATTER_REPOSITORY` is a setting of the end device (or application) in Application Server. When we have a DR client in AS that implements `messageprocessors.PayloadEncodeDecoder` (and it gets registered in [1]), it basically needs to fetch (and cache) the individual formatter. Note that `parameter` will be empty.

There is no need to have flexibility beyond the three codecs, as both Device Repository and The Things Stack are using the three codecs and they are defined in the code as such. No need for arbitrary strings in the API; just define three rpcs.

[1]: https://github.com/TheThingsNetwork/lorawan-stack/blob/53eb523a7fdec61bf1cf172c099fe72de411ad6a/pkg/applicationserver/payload.go#L162

> * The Things Node has invalid profile IDs (mis-match with the filenames), see [TheThingsNetwork/lorawan-devices#25](https://github.com/TheThingsNetwork/lorawan-devices/pull/25)

Fixed

> * [...]

The Things Products was a separate, and now discontinued, company and legal entity.

The Things Network Foundation is LoRa Alliance member and is a separate legal entity too.

neoaggelos, 2020-11-26:
I have started a draft PR with the initial implementation #3530. A few more things that need to be discussed before implementing:

#### Authentication

- Is there any way to verify that a valid access token is being used without checking for specific rights? Would it be simpler if the DR component require the `RIGHT_APPLICATION_DEVICES_WRITE` right (which is needed anyway for the device creation)?

#### Indexing

Some notes on initial bleve usage:

- bleve only supports searching for field values (not keys), so for example it may be tricky to search for devices that support a specific band ID.
- Is simple search enough for us, or do we want to support query string searching?
- We can indeed store and retrieve fields other than the ID. However, fields are stored "flat", and so it is not trivial to load them back from the index. E.g. for an `ttnpb.EndDevice` object, we would have to retrieve every single nested field individually, and then recreate the original proto, which I would rather avoid.

An alternative for that last problem would be to mashal the proto into JSON and then store that under a single key. It works, but it kinda beats the purpose of the index.

#### Photos and videos

I see two options for this. Either add an RPC to serve static files directly, or add some "base URL" configuration, and have the API response prepend that to individual paths. E.g. for `logo.png`, we value we return would be `https://<BASE_URL>/vendor/<vendor>/logo.png`. This should be simple enough to do, and only needs some careful handling for resources that are already URLs (e.g. videos).

#### Payload Formatters

I have added the RPCs to retrieve the individual codecs as suggested. I believe the AS message processor should be implemented in a separate PR, the list of changes is quite large already.

johanstokking, 2020-11-26:
> * Is there any way to verify that a valid access token is being used without checking for specific rights? Would it be simpler if the DR component require the `RIGHT_APPLICATION_DEVICES_WRITE` right (which is needed anyway for the device creation)?

`DEVICES_READ` seems more sane to me, if @htdvisser doesn't have a suggestion for a `RequireAuthenticated()` for general components.

> Some notes on initial bleve usage:

@htdvisser 

> #### Photos and videos
> I see two options for this. Either add an RPC to serve static files directly, or add some "base URL" configuration, and have the API response prepend that to individual paths. [...]

A base URL is fine. Indeed we'll probably synchronize the static files to a public bucket with CDN.

The schema already enforces the photos to be local with specific extensions. Videos are guaranteed to be YouTube and Vimeo urls.

> #### Payload Formatters
> I have added the RPCs to retrieve the individual codecs as suggested. I believe the AS message processor should be implemented in a separate PR, the list of changes is quite large already.

Yes, let's extract that indeed

neoaggelos, 2020-11-30:
Work for store and index and search has started at #3543.

The implementation that has been suggested so far is to:
- Build the index files with CI from the Device Repository and make them available _somewhere_
- TTS retrieves the index files, unzips them and then uses them.
- Index files are used for listing brands and models.
- (WIP) TTS periodically updates the index files. That is, retrieve the new index files, unzip, verify the index is OK, replace old index.

I think that this approach becomes unnecessarily complex, with too many moving parts. Things would probably be simpler if the indexes were a single file instead of a directory. It also requires lots of config, and it will be hard to track down the error in case something breaks.

A simpler alternative method that I believe works here:
- Via CI, create zips of the device repository (obviously, that would not contain any static assets like photos) and make them available somewhere.
- TTS retrieves that archive, unzips locally in a working directory, and then creates the index by itself. Obviously, this takes very little time right now, but I do not think it is going to be an issue in the future either. If this is not considered a terrible idea, it should be simple enough to measure that performance by creating dummy devices.
- Updating is also simpler: Retrieve the new archive, overwrite previous files, and regenerate index.

cc @htdvisser

johanstokking, 2020-11-30:
> * TTS retrieves that archive, unzips locally in a working directory, and then creates the index by itself.

How is this better than fetching and unzipping an index?

neoaggelos, 2020-11-30:
> How is this better than fetching and unzipping an index?

Two main reasons: (a) We don't need any extra logic/process to manage and update the index files (b). Even with an index, we would still need to fetch templates and codecs over HTTP. This is not an issue for templates, but having the codecs in the local filesystem could greatly favor AS performance and protect from possible network connectivity issues.

I know this deviates a bit from the "standard" practice elsewhere in the code (e.g. webhook templates and frequency plans), but then again failure to fetch codecs (which breaks payload decoding) is more critical than failure to fetch webhook templates.

Of course, all of this depends on the fact that creating an index takes a non-significant amount of time.

johanstokking, 2020-12-02:
> > How is this better than fetching and unzipping an index?
> 
> Two main reasons: (a) We don't need any extra logic/process to manage and update the index files (b)

What extra logic?

> Even with an index, we would still need to fetch templates and codecs over HTTP. This is not an issue for templates, but having the codecs in the local filesystem could greatly favor AS performance and protect from possible network connectivity issues.

Sure. I'm thinking more of a "package" that is built on DR CI, that contains the templates and index.

neoaggelos, 2020-12-29:
> What extra logic?

Logic was not the correct word here. I meant having to manage/update/maintain the device repository package file as well.

> Sure. I'm thinking more of a "package" that is built on DR CI, that contains the templates and index.

So, for #3543 I have created a GitHub repository where I build and manage the index file while I'm working on the implementation, but we need to decide where that will be stored eventually.

I have a rough draft of a GitHub workflow for the Device Repository [here](https://github.com/TheThingsNetwork/lorawan-devices/compare/master...neoaggelos:master#diff-5c3fa597431eda03ac3339ae6bf7f05e1a50d6fc7333679ec38e21b337cb6721). The workflow fires when changes are pushed to `master`, creates the Device Repository package and pushes it to a different branch (e.g. `package/v3.11`). The `package/v3.11` branch can be protected, and requires a github token to push.

I have a few other ideas, like adding this as a mage target, dropping it altogether and requiring people to manually fetch and update the device repository package, or even bundling it as part of each TTS release. Thoughts? cc @johanstokking @htdvisser

htdvisser, 2020-12-30:
I think that bundling the latest device repository package with TTS releases is a good idea. We've been bundling https://github.com/TheThingsNetwork/lorawan-frequency-plans and https://github.com/TheThingsNetwork/lorawan-webhook-templates in our AWS AMIs and Docker Images, and that works fine. The only downside is that these packages are now static, in the sense that they only get updated on new releases. For most deployments I think that is perfectly fine, so I'd suggest that for the initial implementation we just focus on just using the bundled repository, and loading that when The Things Stack starts.

In some deployments we may want faster updates of the repositories without the need for updating, or even restarting TTS. For those I think we better wait for Go 1.16 and #3489.

neoaggelos, 2021-01-05:
OK, one solution towards that would be to add the device repository under `/srv/ttn-lorawan/device-repository`. In order to avoid creating issues for development environments, I would prefer having a `SearchPaths []string` config (e.g. defaulting to `[]string{"/srv/ttn-lorawan/device-repository", "device-repository"}`) instead of `WorkingDirectory string`, similar to what we do for static assets with the `public` folder. TTS can then use the first folder it finds.

PR #3530 (2021-01-21): Device Repository

#### Summary

Closes #3433 

#### Changes

- Add new DeviceRepository API and messages under `api/devicerepository.proto`
- Replace `EndDeviceBrand` and `EndDeviceModel` messages and move them to `api/devicerepository.proto`
- Add a new DeviceRepository component
-  Extend list of allowed Field Mask Paths for the new RPCs.
- Add new `BandID` field in IS end devices.
- Everything related to the device repository store (fetching brands/models/templates/codecs) has been extracted to #3543

#### Testing

- Unit testing
- Test locally

##### Regressions

There should not be any

 [... cut]
closed 2021-01-21 by neoaggelos