TheThingsNetwork/lorawan-stack #3442: Add syntax validation for payload formatters

kschiffer, 2020-11-05:
#### Summary
We currently allow any string value to be set as payload formatter. This makes it hard for users to find out about basic syntax issues in their payload formatter scripts.

#### Why do we need this?
- Assisting users better when setting payload formatters
- Avoid easily preventable setting of faulty payload formatters

#### What is already there? What do you see now?
In the console we have some basic syntax highlighting that notifies about issues in the script, however
1. This mechanism only checks against a broad JavaScript syntax, while PFs use ES5 specifically
2. Even when fixing this to use the correct syntax, CLI users will still not have any validation

#### What is missing? What do you want to see?
1. The IS preventing to set PFs with syntax errors
2. The IS responding with a helpful error message that includes the nature and stack trace of the error

#### Environment
v3.10.0

#### How do you propose to implement this?
Haven't worked on this but it's likely not a big issue to pipe the error output of the interpreter to the API response.

#### How do you propose to test this?
Unit tests.

#### Can you do this yourself and submit a Pull Request?
@johanstokking (please reassign as appropriate)

johanstokking, 2020-11-06:
We can probably catch a few things already when evaluating the script. However, we'd need to check all code execution paths to see if the script makes sense. This is already being enforced by the Device Repository.

In any case, catching low hanging fruit errors like entering Lua instead of JavaScript, or using ES6 language features, is easy to catch. Application Server can do that.

trlafleur, 2021-02-05:
yes, this was a great feature in V2, made it easy to test JS prior to deploying...
It allowed sending a test string to the decoder and displaying its output...

johanstokking, 2021-02-10:
I see. I mark this as a duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/2232 then. Please upvote there.

closed 2021-02-10 by johanstokking