TheThingsNetwork/lorawan-stack #3458: Transfer full entity rights to another user on soft-delete of user

pgalic96, 2020-11-10:
#### Summary

When soft-deleting a user, specify another user to transfer rights for entities that do not have another user that has `_ALL` rights for them.

#### Why do we need this?

Prevent having entities that no user has full rights to.

#### What is already there? What do you see now?

User delete logic that simply soft-deletes the user without checking any linked models.

#### What is missing? What do you want to see?

An option in CLI and console to specify another user  which would get full rights to user entities (either all of them or just the ones that do not have another user that has full rights on them).

#### Environment

ttn-lw-stack `v3.10`

#### How do you propose to implement this?

- Expand user delete option to check memberships for all user owned entities
- Transfer rights to the specified user

#### How do you propose to test this?

- Unit tests, manual

#### Can you do this yourself and submit a Pull Request?

Yes I can, right after purge

pgalic96, 2020-11-25:
@johanstokking @htdvisser 

I have one concern with this approach:

- Do we want to have an option where user could specify the receiver of `RIGHTS_ALL` per entity? I can see this as something that could be useful if we want to avoid forcing users to assign the same user to be an owner of all organizations, possibly an organization (or another entity) where they weren't even a collaborator before the delete? On the other hand, this can get quite messy if this needs to be done for a lot of entities.

On the other hand, as this happens only with the entities that the user is the sole owner of, we can put this responsibility on the user, by displaying a warning or something (and they can reassign manually themselves if they want to avoid this situation)

johanstokking, 2020-11-26:
@pgalic96 let's go with one inheriting user for the rights of the deleting user. We may want to come back on this later, but not for now.

pgalic96, 2020-12-02:
@htdvisser I had the following idea:

`Delete` method should not have any parameters in the body (as per `delete` http method), therefore including user id for the transfer would not make sense here.

The solution is to create a new RPC `TransferOrphanedEntities` with user id of the receiver as parameter that would perform this task. It would be performed as part of `delete` CLI command. Then as part of `Delete` RPC we would test if the user to be deleted still has entities that are solely owned, and if this is the case we would output something like `This action will leave following entities orphaned: [entity-ids]. Please re-assign rights to these entities.` This would also mean that the `receiver-id` flag would be optional for the CLI command. Thoughts?

htdvisser, 2020-12-14:
I think it could be useful to have a dedicated RPC to give some user the same rights as another user. Such an RPC could also be useful for admins to quickly copy all rights from one user to another.

I do have some concerns about this, as this would be a quite powerful operation that could do a lot of damage when abused, so I'm not sure we should make this possible before we have 2FA that we can use as an extra confirmation for security.

htdvisser, 2021-03-01:
We've decided to drop this. See https://github.com/TheThingsNetwork/lorawan-stack/pull/3616

closed 2021-03-01 by htdvisser