TheThingsNetwork/lorawan-stack #3469: Fix field styling

bafonins, 2020-11-12:
#### Summary

After https://github.com/TheThingsNetwork/lorawan-stack/pull/3162 the login screen looks like this:
[image]

#### Steps to Reproduce

1. Go to the login screen on the main branch

#### What do you see now?

Broken field  styling

#### What do you want to see instead?

[image]

#### Environment

v3.10

#### Can you do this yourself and submit a Pull Request?

@kschiffer @benolayinka

bafonins, 2020-11-12:
I also noticed that the glossary icon placement varies between desktop and mobile screens:
[image]
[image]

kschiffer, 2020-11-12:
Appears like @benolayinka rebased, force-pushed and merged his outdated local branch instead of the updated one.

bafonins, 2020-11-12:
Additionally, with https://github.com/TheThingsNetwork/lorawan-stack/pull/3162 we no longer can use `cy.findByLabelText` directly. For fields with the glossary icon we must must regex, e.g. instead of `cy.findByLabelText('LoRaWAN Version')`, `cy.findByLabelText(/LoRaWAN Version/)` must be used.

kschiffer, 2020-11-12:
That can be fixed by adjusting the markup and styling a little so that the label content stays pure. However, I don't have time to work on that now.

PR #3474 (2020-11-12): Revert doc link merge commit
#### Summary
This reverts commit 7034f5bcd3feaa3b5c36eb68d3d7000d68815132, reversing
changes made to e8f8b73675a2fc59c221cbf96e3ade96807a6862.

Unfortunately, there were some rebasing issues when the PR was merged (cc @benolayinka). I wasn't able to fix this quickly, so we'll have to revert until we fixed this properly.

Closes #3469

#### Changes
- Revert merge commit

#### Testing

Manual.

##### Regressions

None.

#### Notes for Reviewers
@benolayinka please address the issues and open a new, fixed PR

#### Checklist

- [x] Scope: The referenced issue is addres [... cut]
closed 2020-11-12 by kschiffer