TheThingsNetwork/lorawan-stack #3480: Speed up e2e execution in CI

bafonins, 2020-11-16:
#### Summary

Currently, e2e tests already take ~16minutes to execute (~21mins with several end device e2e tests, see https://github.com/TheThingsNetwork/lorawan-stack/pull/3065). Taking into account that most of the console/oauth app is still lacking any tests. Later with more e2e tests CI pipeline will take even more time. 

#### Why do we need this?

Make sure we do not end up with long CI execution because of e2e tests.

#### What is already there? What do you see now?

A bunch of e2e tests and e2e workflow that executes all tests on almost any PR.

#### What is missing? What do you want to see?

Reduced CI time for e2e tests execution.

#### How do you propose to implement this?

1.
* execute e2e tests in parallel
	- consider [Cypress Dashboard](https://www.cypress.io/dashboard/)
* split e2e tests into groups and run them in separate jobs (e.g. run console, oauth and smoke tests separately)
* Only run smoke tests on pull requests and execute all e2e tests on releases

2. Cache or/and speed up `js:build` target. Also see https://github.com/TheThingsNetwork/lorawan-stack/issues/2661

#### How do you propose to test this?

CI

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2021-02-24:
Consider using an admin API key to perform api calls. See https://github.com/TheThingsNetwork/lorawan-stack/pull/3746

kschiffer, 2021-06-04:
I was able to bring the end-to-end tests execution time down to between <5mins (best case) and 10mins (worst case) by activating parallelization (to 4 machines) and by adding more caching to the workflow runs. An example run can be seen in my fork here:
https://github.com/kschiffer/lorawan-stack/actions/runs/906365351

In order to use parallelization in cypress, we need a [paid account](https://www.cypress.io/pricing). However, it seems like there are cheaper plans for open source projects.

@johanstokking @htdvisser please share your thought about whether you think this is worth pursuing.

johanstokking, 2021-06-09:
The paid account has now been setup.

PR #4264 (2021-06-21): Add frontend caches to release workflow
#### Summary
This quickfix PR adds caching for frontend builds to the snapshot github action workflow. This will speed up builds as well as provide caches for PR builds.

Closes #3480 

#### Changes
- Add public folder cache to `release-snapshot` workflow
- Add babel cache to `release-snapshot` workflow
- Simplify frontend steps in `release-snapshot` workflow
- Simplify cache key for public folder

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with [... cut]
closed 2021-06-21 by kschiffer