TheThingsNetwork/lorawan-stack #3489: Refactor fetch package

johanstokking, 2020-11-17:
#### Summary

Refactor the common `pkg/fetch`.

#### Why do we need this?

To allow for fallback to another source, i.e. when a web server goes down. Another source could be another web location, or a local directory, or any other source.

#### What is already there? What do you see now?

Fetching is currently configured for a single source, and the source itself doesn't support a fallback mechanism.

The HTTP fetcher uses a memory cache transport.

#### What is missing? What do you want to see?

We need to specify multiple alternative HTTP locations that are tried sequentially, randomly or concurrently. We would use this to access geographically distributed or otherwise mirrored data sources that contain the same content.

We also need to be able to specify multiple layers that are tried, for example, first any HTTP source, then a local directory, then a cache with potentially outdated data.

If the HTTP source is configured with multiple URLs, all would fail or time-out in order to try a fallback layer.

Whenever something gets fetched from the primary layer, the cache would be updated.

Keep the compatibility commitment for configuration in mind; all configuration is forward-compatible.

#### How do you propose to test this?

Unit testing with multiple HTTP test servers and local `testdata` dir. We need different scenarios tested.

#### Can you do this yourself and submit a Pull Request?

Can review

neoaggelos, 2020-11-28:
Self-note; let's consider adding a `FileReader(paths ...string) (io.Reader, error)` method, which should improve handling for large files.

johanstokking, 2020-11-30:
> Self-note; let's consider adding a `FileReader(paths ...string) (io.Reader, error)` method, which should improve handling for large files.

What would be the purpose of this?

htdvisser, 2020-12-02:
Before starting work on this, please also read the draft design for the `io.FS` interface that's coming in Go 1.16: https://go.googlesource.com/proposal/+/master/design/draft-iofs.md

The expectation is that we'll soon see several implementations (OS filesystem, HTTP filesystem, S3 filesystem, ...) that implement that interface, so probably smart to go with something similar.