TheThingsNetwork/lorawan-stack #3490: Refactor UDP firewall to filter traffic flood per IP address only

johanstokking, 2020-11-17:
#### Summary

Refactor UDP firewall to filter traffic flood per IP address only

#### Why do we need this?

To block UDP traffic floods before we decode the UDP frame.

#### What is already there? What do you see now?

The current firewall does two things; remote address change detection (to avoid spoofing) and filtering messages when they exceed a configured number over a configured time interval.

#### What is missing? What do you want to see?

We need to take that apart: one firewall for address change detection and one for detecting a message flood and filtering traffic.

The latter would be triggered as soon as an UDP frame has been received, so it only takes remote address into account. The former requires the gateway EUI which is part of the UDP frame, so this stays in place.

#### How do you propose to implement this?

Refactor the UDP firewall, implement in two separate structs. Make sure that configuration stays compatible.

#### How do you propose to test this?

Unit testing

#### Can you do this yourself and submit a Pull Request?

Can review

neoaggelos, 2021-01-11:
The UDP firewall interface deals with `encoding.Packet`, so it is probably not too straightforward to update.

Is it better to add a separate check instead, as part of #1406? (see https://github.com/TheThingsNetwork/lorawan-stack/commit/80e9527c70fad4b38f3fba014d87b3bb005d4642)

johanstokking, 2021-01-11:
> The UDP firewall interface deals with `encoding.Packet`, so it is probably not too straightforward to update.

That is because it currently deals with both the remote IP and the EUI.

> Is it better to add a separate check instead, as part of #1406? (see [80e9527](https://github.com/TheThingsNetwork/lorawan-stack/commit/80e9527c70fad4b38f3fba014d87b3bb005d4642))

That is what is meant with:

> We need to take that apart: one firewall for address change detection and one for detecting a message flood and filtering traffic.

Or do I miss your point here?

johanstokking, 2021-03-01:
Since this relaxes the security of UDP gateways, and can lead to anyone connecting a gateway by EUI, we need #1436.

neoaggelos, 2021-03-30:
Is this already addressed with #3906?

johanstokking, 2021-03-30:
Closed by 36923b8c9f935b51aa6ef6c1ef1368c8cc57cc6b

closed 2021-03-30 by johanstokking