TheThingsNetwork/lorawan-stack #3513: Add support for new RP002 regional parameters specifications

kschiffer, 2020-11-20:
#### Summary
We don't offer support for the [recently released regional parameters specifications](https://lora-alliance.org/search/RP2) `RP002-1.0.0`, `RP002-1.0.1`, `RP002-1.0.2` yet. Notably, these are applicable to all MAC versions. This is especially relevant for LoRaWAN 1.0.4, for which we currently don't provide a fitting PHY version yet (it needs at least `RP002-1.0.1`).

#### Why do we need this?
To support all possible end device certifications.

#### What is already there? What do you see now?
Support for `RP001` specifications (and the coupled one of `TS001`).

#### What is missing? What do you want to see?
Support for the new specifications `RP002-1.0.0`, `RP002-1.0.1`, `RP002-1.0.2`

#### How do you propose to implement this?
Backend
- Add enums to our lorawan protos
- Add definitions to our `pkg/band` package
- Likely some other things

Frontend
- Update PHY version constants in the Console
- Enable selection of new RPs in the end device creation
- Move towards using the official specification name, where applicable

#### How do you propose to test this?
Unit tests.

#### Can you do this yourself and submit a Pull Request?
No.

kschiffer, 2021-03-11:
Also note that we really should adopt the naming of the official specification document `RP002-1.0.0`, `RP002-1.0.1`, `RP002-1.0.2`, as opposed to arbitrary `MAC 1.0.0` `PHY 1.0.0` etc.

johanstokking, 2021-03-11:
Yes, I agree

htdvisser, 2021-04-08:
This is blocking https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/30 and https://github.com/TheThingsNetwork/lorawan-frequency-plans/pull/32

johanstokking, 2021-04-20:
Blocking https://github.com/TheThingsNetwork/lorawan-stack/issues/3806

htdvisser, 2021-04-20:
Blocking https://github.com/TheThingsNetwork/lorawan-frequency-plans/issues/33 as well

johanstokking, 2021-04-20:
@adriansmares would be great if you can pick this up.

@bafonins can do the Console part.

adriansmares, 2021-05-20:
This is currently in progress in https://github.com/TheThingsNetwork/lorawan-stack/tree/feature/3515-rp2-api . The idea is to extend a bit the scope of the issue into fixing, at API level, the naming problems that arise from our `PHYVersion` naming scheme. Both the binary, and the JSON API, are meant to be unaffected by the changes, but I still need to iron out the details (as the changes are invasive).

Moving this to `3.14` as we would like to avoid digging a bigger hole for ourselves by rushing API changes.

bafonins, 2021-07-08:
@adriansmares what is the status here?

adriansmares, 2021-07-13:
The API support has been added in order to allow development to continue in `v3.14`.

I'm going to first do https://github.com/TheThingsNetwork/lorawan-stack/issues/4346 and while working on it add the new RPs.

Blocked by https://github.com/TheThingsNetwork/lorawan-stack/issues/4346.

adriansmares, 2021-10-05:
No longer blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/4346

htdvisser, 2021-12-27:
The v3.16.2 deadline was 2 weeks ago, so moving this to v3.17.0.

htdvisser, 2022-01-05:
Doesn't look like this is going to get finished for v3.17.0 either, so pushing to v3.17.1.

adriansmares, 2022-02-15:
New PHY+MAC pairings that need to be added:
- MAC 1.0.4 has access to all RP2 versions

johanstokking, 2022-02-22:
@kschiffer please reassign to @ryaplots with some instructions. Let's really get this in for 3.18.1. Hopefully this is as easy as it sounds.

closed 2022-03-01 by kschiffer