TheThingsNetwork/lorawan-stack #3515: Revisit RTT calculation for LBS gateways

KrishnaIyer, 2020-11-24:
#### Summary

Revisit RTT calculation for LBS gateways

#### Why do we need this?

We use the LBS `RefTime` parameter returned by the gateway to calculate RTT. This may have issues or the way we are using it is incorrect (under certain conditions).

#### What is already there? What do you see now?

We use the LBS `RefTime` to calculate RTT. It's definition;
> The RefTime field is calculated from the last received MuxTime, adjusted by the time interval on the router between the arrival of MuxTime and the sending of RefTime. Since downlink traffic is less frequent than uplink messages, it is likely that a MuxTime may be reused to construct multiple RefTime fields. This means that round-trip measurements are composed of the latency of the last downlink message and the most recent uplink message.

There are 2 (potential) issues using a time managed by the gateway
1. We don't exactly know what happens in case of infrequent downlinks. Specifically this;
> This means that round-trip measurements are composed of the latency of the last downlink message and the most recent uplink message.

The way I read it, the gateway will internally offset these values and return a sane RefTime but I don't really know.

2. If the gateway's internal clock is faulty/out of sync, it's possible we measure very large values of RTT and hence never use this gateway for downlinks.

#### What is missing? What do you want to see?

That we are able to account for this.

#### Environment

`v3.10 and LBS gateways`

#### How do you propose to implement this?

Should we perhaps Calculate RTT only for TxAcks and use the server time as reference (like UDP)?

#### How do you propose to test this?

UT + Gateway testing

#### Can you do this yourself and submit a Pull Request?

Yup but let's discuss.

johanstokking, 2020-11-24:
I also think that the high RTTs can be caused by a combination of:

1. No downlink for a long time, i.e. the gateway clock drift over hours results in a skewed `RefTime` relative to the server time. With [0,0002% drift](https://doc.sm.tc/station/time.html#clock-synchronization), we'll have one second drift per 1h23m. It could be that some gateway MCUs have more drift, especially under some circumstances (i.e. low temperature)

2. We're never sending `timesync` to the gateway, while [LNS shall do that](https://doc.sm.tc/station/tcproto.html#transferring-gps-time)

---

Proposed solutions:

1. Observe the duration between sending the downlink message (with `MuxTime`) and receiving the uplink message (with `RefTime`) by `Diid`. If this duration is more than some sane limit, don't use it for RTT as the clock may be have drifted too much. We can already observe this duration via functionality that's currently unused, see https://github.com/TheThingsNetwork/lorawan-stack/blob/a1dee5d825a9930e7d427164483a8414605d6e8f/pkg/gatewayserver/io/downlink_token.go#L49-L51

2. Let's start sending `timesync` periodically, i.e. every 30 minutes or so?

KrishnaIyer, 2020-11-24:
Yeah both sound good to me. I'll return with a PR next milestone.

johanstokking, 2020-11-24:
Two more notes based on https://github.com/TheThingsIndustries/lorawan-stack-support/issues/203#issuecomment-732805329:

3. Take only the first `RefTime` after sending a `MuxTime` into account. Subsequent reference times are not relevant and may only add clock drift

4. Make sure that we only take `RefTime` into account _after_ we sent a `MuxTime`. Now, if the `RefTime` is a second or higher, we use it in RTTs, even if we never sent a `MuxTime`. With (3) we probably automatically get this, if we implement it in such a way that setting `MuxTime` sets a flag somewhere that we have to consider `RefTime` once

KrishnaIyer, 2020-11-24:
> Take only the first RefTime after sending a MuxTime into account. Subsequent reference times are not relevant and may only add clock drift

So we could just drop recording RTTs for Uplinks/Joins and instead only do it for TxAcks for the corresponding downlinks. This means that for gateways with little or no downlink, we won't be able to get accurate RTTs. I guess this is better than ridiculous RTT values which prevent downlinks being scheduled.

> Make sure that we only take RefTime into account after we sent a MuxTime. Now, if the RefTime is a second or higher, we use it in RTTs, even if we never sent a MuxTime. With (3) we probably [...]

We actually send the first `MuxTime` on the `RouterConfig` message on every re-connect so this shouldn't be an issue.

johanstokking, 2020-11-24:
> So we could just drop recording RTTs for Uplinks/Joins and instead only do it for TxAcks for the corresponding downlinks

Yes.

> This means that for gateways with little or no downlink, we won't be able to get accurate RTTs

We only need them for downlink. If there are too few RTTs, GS doesn't take them into account, and schedules downlink anyway as the backhaul gets the benefit of the doubt.

> We actually send the first `MuxTime` on the `RouterConfig` message on every re-connect so this shouldn't be an issue.

Does the gateway have some sort of NTP? How does it know the absolute time anyway?

KrishnaIyer, 2020-11-24:
Does the gateway have some sort of NTP? How does it know the absolute time anyway?

The gateway starts with UTC-0 and uses the server sent `MuxTime` for reference.
```
1970-01-01 00:00:39.264 [AIO:DEBU] [2] ws_close reason=1000
1970-01-01 00:00:39.267 [AIO:DEBU] [2] Connection closed unexpectedly
1970-01-01 00:00:39.273 [AIO:DEBU] [2] WS connection shutdown...
1970-01-01 00:00:39.290 [TCE:VERB] Connecting to MUXS...
1970-01-01 00:00:39.743 [TCE:VERB] Connected to MUXS.
2020-11-24 17:45:35.466 [RAL:WARN] Ignoring unsupported/unknown field: antenna_gain
2020-11-24 17:45:35.474 [RAL:INFO] Lora gateway library version: Version: 5.0.1;
2020-11-24 17:45:35.476 [RAL:VERB] Connecting to device:
2020-11-24 17:45:35.478 [RAL:DEBU] SX130x txlut table (0 entries)
2020-11-24 17:45:35.481 [RAL:VERB] SX1301 rxrfchain 0: enable=1 freq=867.5MHz rssi_offset=-166.000000 type=2 tx_enable=1 tx_notch_freq=0
```

PR #3547 (2020-12-08): Fix RTT handling for LBS gateways

#### Summary

Closes #3515 

Moved `timesync` message support to #3546 

#### Changes

- Measure RTT only for valid TxAcks.
- Add sanity check for round trip time
- Update tests

#### Testing

UT (testing on gateway ongoing)

##### Regressions

Traffic test on physical gateways should cover regressions.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README. [... cut]
closed 2020-12-08 by KrishnaIyer