TheThingsNetwork/lorawan-stack #3516: Non-interactive ttn-lw-cli login

htdvisser, 2020-11-24:
#### Summary

The `ttn-lw-cli login` command is currently always interactive. For scripted deployments however, it could be useful to disable interaction, and use flags for everything.

#### What is already there? What do you see now?

- `ttn-lw-cli login` always tries to open a browser window
- `ttn-lw-cli login` does not accept the authorization code in a flag when using `--callback=false`

#### What is missing? What do you want to see?

- The command should get a `--interactive=false` flag
- The command should get a `--authorization-code` flag

neoaggelos, 2020-11-25:
Related #2894

htdvisser, 2020-11-26:
Thanks for linking that, @neoaggelos.

After giving this some more thought, I think it's better to do #3518 instead of this one. With #3518 we can do non-interactive CLI logins with the existing `ttn-lw-cli login --api-key XXX`, and we can keep the OAuth flow interactive, as it should be.

I really don't want to accept username/password as suggested in #2894, so I'll comment and close that issue too.

closed 2020-11-26 by htdvisser