TheThingsNetwork/lorawan-stack #3517: Time when devices are "created" are not updated

laurensslats, 2020-11-24:
#### Summary
See the 2 devices which show `Now` below the `Created` column.
[image]

#### Steps to Reproduce

1. Look at the application in TTIs tenant (only access for admins) - [link](https://tti.eu1.cloud.thethings.industries/console/applications/smart-campervan/devices)
2. OR, create a new devices, and watch the device overview page.

#### What do you see now?
See img above.

#### What do you want to see instead?
The created-field being updated.

#### Environment
v3.10

#### How do you propose to test this?
Add new devices locally

laurensslats, 2020-11-24:
Note.

This doesn't happen when adding new devices. It only happend with 2 devices I added in the past.

See:
[image]

asmulko, 2020-12-29:
@bafonins @kschiffer unfortunately PR #3588 did not solve the issue. Could someone please point me in the right direction or perhaps take a look at this issue themselves?

kschiffer, 2021-06-01:
I think we can close this since the creation date is not shown anymore in the device list (via #3832) . Also we haven't been able to reproduce the issue.

closed 2021-06-01 by kschiffer