TheThingsNetwork/lorawan-stack #3549: Clean Up Old Docker Snapshots

htdvisser, 2020-11-30:
#### Summary

We should periodically clean up old Docker snapshots

#### Why do we need this?

Because we don't need to keep every snapshot forever

#### What is already there? What do you see now?

Every merge to a minor branch (such as `v3.10`) triggers a snapshot build, and pushes an image to `lorawan-stack-dev:<git hash>`. These tags accumulate over time.

#### What is missing? What do you want to see?

We should periodically run a script to delete snapshot tags older than a couple of weeks.

#### How do you propose to implement this?

On Docker Hub:

- List images (https://docs.docker.com/registry/spec/api/#listing-image-tags)
- Delete images (https://docs.docker.com/registry/spec/api/#deleting-an-image)

GitHub Container Registry currently does not support deleting images through the API (https://docs.github.com/en/free-pro-team@latest/packages/managing-container-images-with-github-container-registry/deleting-a-container-image#about-package-deletion), but I assume that will be supported in the future.

htdvisser, 2021-04-01:
Just did another manual cleanup (but this time from Retool): https://thethingsindustries.retool.com/apps/Docker%20Images

Perhaps we can take that as a starting point for the cleanup tool.

htdvisser, 2022-07-28:
This may be a nice one for @happyRip to pick up, perhaps with help from @KrishnaIyer or @adriansmares.

happyRip, 2022-07-28:
The workflow seems pretty simple. You see this as a `shell` or `go` tool?

KrishnaIyer, 2023-07-04:
Moved to infrastructure;

closed 2023-07-04 by KrishnaIyer