TheThingsNetwork/lorawan-stack #3579: Push "latest" Docker image from release workflow

htdvisser, 2020-12-09:
#### Summary

Instead of manually updating the `latest` tag for our Docker images, this should be done by CI.

#### What is already there? What do you see now?

A step in the release template that says to manually update the `latest` tag.

#### What is missing? What do you want to see?

CI automatically doing this if we're doing the most recent release.

#### Environment

GitHub Actions

#### How do you propose to implement this?

- List the git tags present in the repository
- If the current build is for the latest version number, also push the `latest` tag

nejraselimovic, 2021-06-07:
@KrishnaIyer you can pick this one up, right? This would be beneficial for the release process that we're trying to improve, so I'll schedule this for Q3.

htdvisser, 2021-06-08:
Or maybe @kschiffer can pull this into the work he's already doing on our CI workflows?

KrishnaIyer, 2022-03-23:
Since @michalborkowski96 worked on this, can you take a look and see if something needs to be done here?

michalborkowski96, 2022-03-23:
https://github.com/TheThingsNetwork/lorawan-stack/blob/b053b3b10089d5c9be0ad324fa9db78a0659ea97/.github/workflows/release-tag.yml#L145

closed 2022-03-23 by michalborkowski96