TheThingsNetwork/lorawan-stack #3585: Allow hard-delete

bafonins, 2020-12-10:
#### Summary

With https://github.com/TheThingsNetwork/lorawan-stack/pull/3467 we allow deleting entities and releasing their ids. We should add this to the Console.

#### Why do we need this?

Allow the user to chose between soft and hard deletion.

#### What is already there? What do you see now?

Delete modal
[image]

that uses `Delete` RPC's for each entity.

#### What is missing? What do you want to see?

Option to perform hard deletion.

#### Environment

v3.10.4

#### How do you propose to implement this?

IMO, we can extend the existing modal with `Release entity ID` checkbox that initially is unchecked.
cc @kschiffer 

#### How do you propose to test this?

e2e tests and manual testing

#### Can you do this yourself and submit a Pull Request?

yes

kschiffer, 2020-12-10:
Here's how I think this should work:

1.  The delete action in the Console should always trigger the purge endpoint.
2.  In case the application is "pristine", e.g. has no devices, and other entities associated whatsoever, we can keep the modal behavior as it is right now.
3.  In case the application is not "pristine", the modal should be updated informing that all devices, webhooks, API keys and other data associated with the application are deleted irretrievably. Additionally, for such high-danger actions, we should add a "type name of entity" confirmation (similar to how deleting repository works on Github).

johanstokking, 2021-02-11:
References #3739

kschiffer, 2021-05-04:
I Suggest to reconsider the milestone/prioritization since we now release entity IDs automatically when deleting via the Console. I think it's a good idea now to update the deletion flow as mentioned in my comment above.

Adding to 3.:

We should make a typed confirmation for applications, gateways and users/organizations.

htdvisser, 2021-06-02:
Pushing this to v3.13.2. I don't think @bafonins is going to be working on this on his day off, so this definitely won't make it to today's deadline for v3.13.1.

htdvisser, 2021-06-24:
Re-opening this, since I don't think the permanent delete should be the default for admins. Security/privacy guarantees go out of the window when releasing entity IDs. If someone else registers a deleted ID, they may suddenly have access to stored data of the old entity. This is not something we can fix without breaking V3 compatibility, so we should try not to release IDs until V4 at least.

I propose that we change this to let admins toggle between the default soft-delete and the permanent delete.

htdvisser, 2021-06-24:
@bafonins please put this in an appropriate milestone, depending on your availability.

bafonins, 2021-06-24:
@kschiffer What should be changed in the UI?

kschiffer, 2021-06-24:
Let's have two buttons then for admins next to each other then `Soft delete {entity}` and `Hard delete {entity}`.

kschiffer, 2021-07-15:
@htdvisser so as far as I understand, we can provide this option for admin users when deleting an entity, but do not make it a default. Basically it would mean adding a checkbox to the deletion modal that would change the operation to purge, together with a tooltip explaining the implications for security:

```
Releasing the entity IDs will make it possible to register a new entity with the same ID. Note that this could potentially generate security issues when  such device is accidentally or purposefully created from a third party, since it will effectively take-over the security context of the deleted device.
```

Please confirm.

htdvisser, 2021-07-15:
Originally the idea was that "purge" would only be available for deleted entities (in the Console: from the "list deleted" views), and only after a some cooldown period, but that idea got lost along the way.

I think we need to be even more clear about what the "purge" checkbox does so that users understand what is (and isn't) going to happen.

Purging an (entity)...

- will **permanently** delete the (entity)
- will **permanently** delete collaborator rights related to the entity (does not apply to users, those don't have collaborators)
- will **permanently** delete API keys related to the entity (does not apply to OAuth clients, those don't have API keys)
- will **not** delete stored events for the (entity)
- will **not** delete information stored in Packet Broker (applies to gateways)
- will **not** delete stored data in the storage integration (applies to applications)
- will **not** delete stored data in external integrations (applies to applications)
- will release the ID of the (entity) for re-use, which **may give other users access to historical data** of the (entity)

kschiffer, 2021-07-15:
Ok so let's do the following:

*   Add the purge checkbox for admins as discussed above
*   Add a tooltip that also links to a (new) page in the docs that explains the implications of purging, cc @benolayinka 
*   Let's go full circle and add that initial idea in the entity tables where there will be a new tab (visible for admins only) that shows deleted entities and displays a purge button on the table row to purge individual entities.

bafonins, 2021-07-28:
Reopening this issue to keep track of the deleted entity tables as mentioned in https://github.com/TheThingsNetwork/lorawan-stack/issues/3585#issuecomment-880683737

benolayinka, 2021-08-05:
How is hard delete done in the console now? I'd like to add a screenshot to https://www.thethingsindustries.com/docs/reference/purge/

bafonins, 2021-08-10:
@benolayinka Currently, entity deletion works as follows:

[image]
[image]

1. If the user is admin then
* there is an option to perform hard deletion (can reuse entity id)
* use regular soft deletion (cannot reuse entity id)
2. If the user is not admin, then there only one option: regular soft deletion

Also, see screenshots

closed 2021-08-24 by bafonins