TheThingsNetwork/lorawan-stack #3602: Rename JS SDK package to `tts`

kschiffer, 2020-12-17:
#### Summary
Inline with our rebranding, we should also rename the package name of our JS SDK, which is currently named `ttn-lw`.

#### Why do we need this?
Consistent naming and branding.

#### What is already there? What do you see now?
`ttn-lw` package name.

#### What is missing? What do you want to see?
- Renaming to `tts`
- Claiming `tts` name in NPM as well
- Some respective updating of tooling and imports

#### How do you propose to implement this?
Trivial.

#### How do you propose to test this?
Manual.

#### Can you do this yourself and submit a Pull Request?
Yes.

benolayinka, 2021-02-24:
@kschiffer this is keeping the rebranding umbrella issue open. Do you have a timeline for it?

kschiffer, 2021-02-24:
Commented in the other issue. I don't think we should keep the umbrella issue open because of this, since this is quite tangential.