TheThingsNetwork/lorawan-stack #3606: End-to-end test coverage

kschiffer, 2020-12-21:
#### Summary
This umbrella issue keeps track of our end-to-end test coverage.

* [ ]   Applications
    * [ ]   CRUD
      * [x] Create
      * [ ] Read
      * [ ] Update
      * [ ] Delete
    * [ ]   Checking overview (information displayed as expected)
    * [ ]   Linking (set and remove)
    * [ ]   Payload formatters (set all types, remove)
    * [ ]   Integrations
        * [ ]   MQTT (Check credentials)
        * [ ]   Webhooks (CRUD, with and without webhook template)
        * [ ]   Pub/sub (CRUD)
          * [x] Create
          * [ ] Read
          * [ ] Update
          * [ ] Delete
    * [ ]   Collaborators (CRUD)
    * [ ]   API Keys (CRUD)
* [ ]   End devices
    * [ ]   CRUD
      * [x] Create
      * [ ] Read
      * [ ] Update
      * [ ] Delete
    * [ ]   Import
    * [ ]   Checking overview (information displayed as expected)
    * [ ]   Scheduling downlinks
    * [ ]   Location (setting and removing)
    * [ ]   Payload formatters (set types, remove, check precedence over app level PFs)
    * [ ]   Claim authentication code (setting and removing)
    * [ ]   General Settings (test various sub-forms, removing gateway)
* [ ]   Gateways
    * [ ]   CRUD
      * [ ] Create
      * [ ] Read
      * [x] Update
      * [x] Delete
    * [ ]   Checking overview (information displayed as expected)
    * [ ]   Location (setting and removing, confirm updating via status message)
    * [ ]   Collaborators (CRUD)
    * [ ]   API Keys (CRUD)
* [ ]   Organizations
    * [ ]   CRUD
      * [ ] Create
      * [ ] Read
      * [ ] Update
      * [ ] Delete
    * [ ]   Checking overview (information displayed as expected)
    * [ ]   Collaborators (CRUD)
    * [ ]   API Keys (CRUD)
* [ ] Authentication and authorization
  * [x] Login
  * [ ] Logout (Console and OAuth app)
  * [x] User registration
  * [x] Change password (when authenticated)
  * [ ] Change password (using temporary password)
  * [ ] Authorizing an OAuth client
  * [ ] Verifying contact info
  * [ ] Retrieving authorization code (`/oauth/code`)
* [ ]   Miscellaneous
    * [ ]   Feature toggles
        * [ ]   Check various rights setups and confirm:
            * [ ]   Correct visibility of items
            * [ ]   Correct write restriction of items
    * [ ]   Data / Event streams (all entities)
        * [ ]   we need some work to enabling simulating events from within the cypress test
        * [ ]   Confirm correct rendering of various event types
        * [ ]   Test pausing and clearing
    * [ ]   User management (CRUD)
    * [ ]   Search functionality in tables
    * [ ]   Some smoke testing of responsive styles
* [ ] Add functional tests to `storybook`  #3478
* [ ] Add page title assertions to e2e tests #3479
* [ ] Speed up e2e execution in CI #3480
* [ ] Add clipboard support for e2e tests #3481
* [ ] Add direct database access for end-to-end tests #3605
* [ ] Save screenshots as artifacts for failed e2e tests #3523

Everyone please keep this up to date and add test-cases that you think are important, or split up cases into more detailed sub-cases.

For everyone that wants to dive into writing end-to-end tests, please see the guide: https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.10/DEVELOPMENT.md#end-to-end-tests

kschiffer, 2020-12-22:
I'll add the rest of the dev team as assignees once #3611 is done.

asmulko, 2021-01-10:
@kschiffer I assume that `Application / Linking` e2e tests can be removed from the list, because it will be removed in v3.11, see #3614 and we can put `Payload formatters` on hold as well

kschiffer, 2021-01-18:
Yes indeed. Linking can be removed though `Payload formatters` should stay untouched as far as I understand.

johanstokking, 2021-02-11:
What is the status here? What can we improve on the short term still?

kschiffer, 2021-02-11:
@asmulko made a lot of progress with this recently and I think the coverage is pretty good already. The most important thing short-term is traffic simulation, which will enable us putting a whole range of critical end-to-end tests in place. This is also tracked in #3611. Then we need to start looking into performance optimizations [#3480](https://github.com/TheThingsNetwork/lorawan-stack/issues/3480).

johanstokking, 2021-02-12:
OK. Should we track end-to-end testing in this issue or in #3611, or is there a reason to keep both?

kschiffer, 2021-02-15:
I'd prefer keeping and maintaining this overview here as well. We can update this issue with a lot more test cases and onboard more of the team to write end-to-end tests after [#3611](https://github.com/TheThingsNetwork/lorawan-stack/issues/3611) is done.

kschiffer, 2021-11-10:
Little reminder for me, @bafonins, and @ryaplots to keep the list above updated and to periodically check back whether some of those can be picked up by us.

kschiffer, 2023-11-09:
I think we an close this now. The rest of the missing coverage is either tracked by individual issues or not planned currently.

closed 2023-11-09 by kschiffer