TheThingsNetwork/lorawan-stack #3624: Redirect to /console after log in to TTS community edition

nejraselimovic, 2020-12-29:
#### Summary
When I log in to the eu1 tenant on our new community cluster, I get redirected to [https://eu1.cloud.thethings.network/oauth/?](https://eu1.cloud.thethings.network/oauth/?) and there is no way to move forward to the Console. 

#### What do you see now?
[image]

#### What do you want to see instead?
A redirection to [https://eu1.cloud.thethings.network/console/](https://eu1.cloud.thethings.network/console/).

#### Can you do this yourself and submit a Pull Request?
Nope

@laurensslats for now, this is the only thing I spotted while going through the onboarding process

nejraselimovic, 2020-12-30:
@kschiffer is this something you're covering?

kschiffer, 2020-12-30:
Yes. It will be part of the Account App, though I will branch this fix off today to make it available already.

closed 2021-01-07 by nejraselimovic