TheThingsNetwork/lorawan-stack #3632: Coding Rate left out if first gateway in rx_metadata is packetbroker

jpmeijers, 2021-01-03:
#### Summary

Sometimes the Coding Rate field is not reported in the json received via MQTT. I have the suspicion it has to do with the Packet Broker. If the packet is received from the PB first, the CR is left out. If the packet is received directly form a gateway first, the CR is added. 

#### Steps to Reproduce

1. Connect a gateway to TTN. 
2. Connect another gateway to a TTI cloud instance. 
3. Register a device on the TTI cloud instance and let it send periodic uplinks.
4. Make sure the device is received by both gateways.
5. Subscribe to uplink messages on MQTT.
6. Keep an eye on the `rx_metadata` and the `settings` object.
7. When `packetbroker` is the first gateway in `rx_metadata` there is no `coding_rate` in the `settings`.

#### What do you see now?

Two examples of the same network, tenant, gateways and device attached. Logs captured a few minutes apart.
[tti-cr-uplink.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/5761881/tti-cr-uplink.txt)
[tti-no-cr-uplink.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/5761882/tti-no-cr-uplink.txt)

```
      "settings":{
         "data_rate":{
            "lora":{
               "bandwidth":125000,
               "spreading_factor":7
            }
         },
         "data_rate_index":5,
         "frequency":"867100000"
      },
```

#### What do you want to see instead?

```
      "settings":{
         "data_rate":{
            "lora":{
               "bandwidth":125000,
               "spreading_factor":7
            }
         },
         "data_rate_index":5,
         "coding_rate":"4/5",
         "frequency":"867300000",
         "timestamp":2327339579
      },
```

#### Environment

eu1.cloud.thethings.industries

v3.10.5

#### How do you propose to implement this?

Find the reason why the CR is left out sometimes.

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

No

neoaggelos, 2021-01-03:
References https://github.com/packetbroker/api/issues/21

rvolosatovs, 2021-01-04:
This can easily be made part of deduplication in NS (note, @adriansmares reported this to me some time ago, cc for additional insights).
@johanstokking do we expect coding rate to be set in PBA at all times? If not:
1. Shall NS try to determine coding rate during deduplication?
2. Are there other parameters not present in `RXMetadata`, which NS may need to merge?

johanstokking, 2021-01-05:
Already fixed with https://github.com/TheThingsNetwork/lorawan-stack/pull/3600

This is landing in 3.10.6

closed 2021-01-05 by johanstokking

johanstokking, 2021-01-05:
> @johanstokking do we expect coding rate to be set in PBA at all times?

Yes

> 1. Shall NS try to determine coding rate during deduplication?

We can, but that's really goldplating, as you can expect it from PB since 3.10.6

> 2. Are there other parameters not present in `RXMetadata`, which NS may need to merge?

Not that I know of