TheThingsNetwork/lorawan-stack #3633: Variable type error

wong-hao, 2021-01-04:
#### Summary

[Temperature of the Gateway](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.10/pkg/ttnpb/udp/packet_data.go#L118) is not **signed integer**, it is [float](https://github.com/wong-hao/lora_gateway/blob/master/subtree/PROTOCOL.md).
...

#### Steps to Reproduce

1. ...
2. ...
3. ...

#### What do you see now?

...

#### What do you want to see instead?

...

#### Environment

...

#### How do you propose to implement this?

Just change the type.
...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

KrishnaIyer, 2021-02-05:
https://github.com/Lora-net/sx1302_hal/blob/master/packet_forwarder/PROTOCOL.md
This is the `SX1302` reference.

`temp` field isn't defined in the [`SX1301`](https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT) reference.

This field must've been added somewhere along the way? @johanstokking maybe you have some context here.

> Just change the variable type here.

We can't simply edit gateway facing fields. Check our [compatibility commitment](https://github.com/TheThingsNetwork/lorawan-stack#commitments-and-releases) for more details.

We'd probably need to able to check if a gateway uses `SX1301` or the `SX1302` ref (added to the `Gateway` proto) and refactor the translation. This isn't a simple change. We can discuss this now and move  to `Next Up`.

wong-hao, 2021-02-05:
> https://github.com/Lora-net/sx1302_hal/blob/master/packet_forwarder/PROTOCOL.md
> 这是`SX1302`参考。
> 
> `temp`[`SX1301`](https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT)引用中未定义该字段。
> 
> 必须在此过程中的某个位置添加此字段吗？@johanstokking也许您在这里有一定的了解。
> 
> > 只需在此处更改变量类型。
> 
> 我们不能简单地编辑面向网关的字段。查看我们的[兼容性承诺](https://github.com/TheThingsNetwork/lorawan-stack#commitments-and-releases)以获取更多详细信息。
> 
> 我们可能需要检查网关是否使用`SX1301`了`SX1302`ref或ref（添加到`Gateway`proto）并重构了翻译。这不是一个简单的更改。我们现在可以讨论这个，然后移至`Next Up`。

I can provide an experimental phenomenon for your analysis
When I use sx1302, if I don't handle this error, TTN console can't receive my message. 
When I comment out the temperature field in the **semtech UDP packet forwarder code** (let the uploaded data not contain the temperature attribute, which is exactly the same as sx1301), TTN can receive the gateway message.

wong-hao, 2021-02-05:
> https://github.com/Lora-net/sx1302_hal/blob/master/packet_forwarder/PROTOCOL.md
> This is the `SX1302` reference.
>
> `temp` field isn't defined in the [...]

I can provide an experimental phenomenon for your analysis
When I use sx1302, if I don't handle this error, TTN console can't receive my message.
When I comment out the temperature field in the semtech UDP packet forwarder code (let the uploaded data not contain the temperature attribute, which is exactly the same as sx1301), TTN can receive the gateway message.

johanstokking, 2021-02-10:
> > Just change the variable type here.
> 
> We can't simply edit gateway facing fields. Check our [compatibility commitment](https://github.com/TheThingsNetwork/lorawan-stack#commitments-and-releases) for more details.

We aren't sending these values back to the gateway.

It's safe to change this to `*float32`.

PR #3826 (2021-02-18): Change type of temp field to float32 in UDP stats message

#### Summary

Closes #3633

#### Changes

- Change field type
-  Update tests

#### Testing

UT

##### Regressions

I don't expect any since this field is actually used as `float32` when [translated to metrics](https://github.com/TheThingsNetwork/lorawan-stack/blob/04b4523d36a4f2505cdbfa158b60084300a57aa1/pkg/ttnpb/udp/translation.go#L227).

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration a [... cut]
closed 2021-02-18 by KrishnaIyer