TheThingsNetwork/lorawan-stack #3637: Push API to Device Class C End Device blocked by Network Server with FCnt too low error

dpostorivo, 2021-01-04:
#### Summary

On several Device Class C devices in the field we are seeing an error when we try to push down link messages via API. This seems to completely block all down messages until the device leaves then rejoins the network. This seems to be tracked internally to the ttn stack, and the API has no reference to is in the push API.

#### Steps to Reproduce

1. Have a class C device join.
2. Send messages to the device until FCnt (typically around 200-300)

#### What do you see now?
Once the things stack, all down links to the device gives a similar error to below:

```
error:pkg/networkserver:f_cnt_too_low (FCnt `171` is lower than minimum of `173`)
```
#### What do you want to see instead?

Down links should still go though and should not be blocked or there should be a reasonable recovery path.

#### Environment

End device is an XDot in class C mode version: 3.2.5

Gateway is a Sentrius RG1xx gateway version: Laird Linux gatwick-laird-93.8.5.21

ttn-lw-stack version:  3.10.4

#### How do you propose to implement this?

Either track down why the FCnt is being flagged internally as low or have an internal recovery path.

#### How do you propose to test this?

We have a test environment where we are currently seeing on and can deploy fixes to it to see if the devices recover.

#### Can you do this yourself and submit a Pull Request?

I'd like to, but I'm not as familiar with this codebase and would need guidance/review to make sure the fix makes sense for all users.

adriansmares, 2021-01-08:
Can you please check your stack logs for errors such as the one bellow, and provide the context you have there (logs above / logs bellow) ? 
```
  WARN Failed to recalculate downlink queue; clear the downlink queue device_uid=app1.dev2 error=error:pkg/networkserver:f_cnt_too_low (FCnt `5` is lower than minimum of `9`) f_cnt=5 grpc.method=HandleUplink grpc.service=ttn.lorawan.v3.NsAs min_f_cnt=9 namespace=grpc peer.address=pipe request_id=01EVHB1Y37M4VZQG9NSBMPYQJ0 session_key_id=0176E2AF8F3EE9B4D70F6622ABF68ACF
```
They should occur _before_ the AS stops accepting downlinks. We need more context on how the sessions actually end up out-of-sync.

dpostorivo, 2021-01-11:
Still combing through our logs to find exactly when the first occurrence happens, but could not find any occurrences of  `Failed to recalculate downlink queue` as of now. We're setting up more devices to help identify the problem, so hopefully I'll have more useful logs soon.

Is there any other log messages I should be scanning for?

adriansmares, 2021-01-12:
> Still combing through our logs to find exactly when the first occurrence happens, but could not find any occurrences of `Failed to recalculate downlink queue` as of now. We're setting up more [...]

Unfortunately none that I can think of. 

@rvolosatovs maybe we should log with `DEBUG` when a `DownlinkQueueInvalidated` uplink is sent to the AS (basically when you drain `ifScheduledApplicationUps`) ? We can then log in the AS when such an uplink is received and check in the logs if things stay in sync in our communication protocol. _Maybe_ somehow we're losing invalidations somewhere.

rvolosatovs, 2021-01-12:
> Unfortunately none that I can think of.
>
> @rvolosatovs maybe we should log with `DEBUG` when a `DownlinkQueueInvalidated` uplink is sent to the AS (basically when you drain [...]

Filed https://github.com/TheThingsNetwork/lorawan-stack/issues/3658 for this

johanstokking, 2021-03-01:
@adriansmares what is the status here?

adriansmares, 2021-03-01:
To be fixed as part of https://github.com/TheThingsNetwork/lorawan-stack/issues/3649 .

closed 2021-03-03 by adriansmares