TheThingsNetwork/lorawan-stack #3686: Support dynamic class B ping slot frequency

johanstokking, 2021-01-20:
#### Summary

Support dynamic class B ping slot frequency.

References https://github.com/TheThingsNetwork/lorawan-devices/pull/38#discussion_r555044700

#### Steps to Reproduce

Onboard a device with, for example, band CN470 that supports class B.

#### What do you see now?

There's no fixed ping slot frequency, which is a required value by NS. It is not possible to specify empty;

```json
{
  "code": 3,
  "message": "error:pkg/networkserver:field_mask (invalid field mask)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/networkserver",
      "name": "field_mask",
      "message_format": "invalid field mask",
      "correlation_id": "2b2b2538c031439d899f8793d604b3f8",
      "cause": {
        "namespace": "pkg/ttnpb",
        "name": "missing_field",
        "message_format": "field `{field}` is missing",
        "attributes": {
          "field": "mac_settings.ping_slot_frequency.value"
        },
        "code": 2
      },
      "code": 3
    }
  ]
}
```

#### What do you want to see instead?

This should be dynamic and depending on the band.

#### How do you propose to implement this?

Support dynamic ping slot frequency as defined in LoRaWAN Regional Parameters based on the band; CN470, US915 and AU915 seem to be affected.

#### How do you propose to test this?

Test creating devices.

#### Can you do this yourself and submit a Pull Request?

Can review

rvolosatovs, 2021-01-20:
Indeed, didn't see this in the spec before.
We'll have to do some refactoring to allow for this.

Also, how would this work for ABP? I suppose we will still have to force `mac_settings.ping_slot_frequency.value` to be set for those devices

johanstokking, 2021-01-21:
> Also, how would this work for ABP? I suppose we will still have to force `mac_settings.ping_slot_frequency.value` to be set for those devices

I guess so, yes.

janakj, 2022-04-04:
I just came across what I believe to might be the same or similar issue. I am in US915 and when I enable class B for my device and configure beacon and ping slot frequency to 923.3 MHz manually, I get the following error:

[image]

Is there any way to enable class B in US915 with TTN v3.18.2?

closed 2022-08-31 by adriansmares