TheThingsNetwork/lorawan-stack #3705: Expose tenant id in webhook tempalte

jpmeijers, 2021-01-25:
#### Summary

When a webhook sends data to a third party service, the json from `tenantA.eu1.cloud.thethings.industries`, `tenantB.eu1.cloud.thethings.industries` and `eu1.cloud.thethings.network` looks the same. The third party service therefore has no way to know which LoRaWAN network was the source for this data. One can guess from the source IP address, but an IP can change, and that won't differentiate between multiple tenants on the same instance.

#### Why do we need this?

Identify the network that was used to forward the packet.

#### What is already there? What do you see now?

The application id and device id, along with gateway id if the packet was not forwarded by the packet broker.

The webhook template does not seem to have contextual variables that can be used and set in the headers.

#### What is missing? What do you want to see?

Tenant id at least, but if we can get the full server hostname, that would be even better.

#### Environment

The Things Stack v3.10.5 and v3.11.0-rc1

#### How do you propose to implement this?

Expose the tenant id (or better the full hostname) as a variable in the webhook template, or always add it as header along with a webhook post.

#### How do you propose to test this?

Add a webhook template and see if the network that was the source for this packet can be identified.

#### Can you do this yourself and submit a Pull Request?

No

jpmeijers, 2021-02-13:
I was looking into this issue again today, and after reading https://github.com/TheThingsNetwork/lorawan-stack/pull/1011 I considered adding TenantID and ClusterID to `expandVariables()` in `webhooks.go`. Unfortunately I do not understand how to get the Tenant and Cluster IDs from the context.

https://github.com/TheThingsNetwork/lorawan-stack/pull/1011/files#diff-91a1940fc428bf2d3f8b28f3bde1e3dbf817b39c21b8d35d2276ad5d2e8ce3f0R414

While looking at the code I thought, why don't we just add the tenant and cluster IDs standard into the `end_device_ids`? Technically the cluster and the tenant in that cluster is part of the unique end device ID.

jpmeijers, 2021-02-13:
Another workaround I am looking at is the downlink headers. I am not sure if we can use a webhook template to always include these headers, even when the api keys is not set. But it can work.

Example headers from TTNv3:
```
array (
  'HOST' => 'ttnmapper.org',
  'ACCEPT-ENCODING' => 'gzip',
  'X-DOWNLINK-PUSH' => 'https://eu1.cloud.thethings.network/api/v3/as/applications/jpm-test/webhooks/post-to-ttnmapper/devices/eui-0004a30b001c684f/down/push',
  'X-DOWNLINK-APIKEY' => 'NNSXS.redacted',
  'USER-AGENT' => 'ttn-lw-application-server/3.11.0-rc2',
  'X-DOWNLINK-REPLACE' => 'https://eu1.cloud.thethings.network/api/v3/as/applications/jpm-test/webhooks/post-to-ttnmapper/devices/eui-0004a30b001c684f/down/replace',
  'AUTHORIZATION' => '',
)

```

Example headers from cloud hosted:
```
array (
  'HOST' => 'ttnmapper.org',
  'ACCEPT-ENCODING' => 'gzip',
  'USER-AGENT' => 'ttn-lw-application-server/3.10.9',
  'X-DOWNLINK-REPLACE' => 'https://the-box.eu1.cloud.thethings.industries/api/v3/as/applications/test-app/webhooks/ttnmapper-test/devices/thingsuno/down/replace',
  'X-DOWNLINK-PUSH' => 'https://the-box.eu1.cloud.thethings.industries/api/v3/as/applications/test-app/webhooks/ttnmapper-test/devices/thingsuno/down/push',
  'X-DOWNLINK-APIKEY' => 'NNSXS.redacted',
  'AUTHORIZATION' => '',
)
```

Using the X-DOWNLINK-PUSH or the X-DOWNLINK-REPLACE headers, one can uniquely identify the LoRaWAN server instance that handled the uplink message. That is because this URL contains the TenantID and the ClusterID, but mainly because this URL is unique per tenant and any private or public instance for the stack.

@htdvisser what is your opinion? Do you think I can rely on these headers? And more importantly, can I somehow craft my webhook template to always include these headers?

htdvisser, 2021-02-15:
While you _can_ use those headers (right now), I don't think you _should_ (in the long term).

Much better would be if your users register the necessary parameters in your application, and then you give them some token (that identifies those parameters) and let them configure that in the webhook template, so that the token is included in every request.

jpmeijers, 2021-02-17:
I'll close this issue, as the method discussed in https://github.com/TheThingsNetwork/lorawan-stack/issues/3795 is a better approach.

closed 2021-02-17 by jpmeijers