TheThingsNetwork/lorawan-stack #3707: Auto fill device id if it's blank and an EUI is specified

jpmeijers, 2021-01-27:
#### Summary

When one registers a device, automatically fill the device ID with `eui-abcdef0123456789` when the EUI (`ABCDEF012345678`) is filled in but the ID field is still blank.

Having more than one immutable identifier for a single physical thing can become confusing. Therefore I stick to this scheme to identify devices. The ID and the EUI are therefore the same. I then leave the name and description fields blank.

#### Why do we need this?

Make registering devices quicker.

#### What is already there? What do you see now?

I have to manually type in the ID in the required format.

#### What is missing? What do you want to see?

Automatically fill the ID field with the eui-... format.

#### Environment

TTS v3.11.0-rc1

#### How do you propose to implement this?

Maybe some javascript on the console?

#### How do you propose to test this?

Register a device on the console and see if the ID gets filled in automatically.

#### Can you do this yourself and submit a Pull Request?

No

kschiffer, 2021-01-28:
References #3343, the wireframe there already includes this functionality.

kschiffer, 2021-03-11:
Resolved via https://github.com/TheThingsNetwork/lorawan-stack/pull/3801

closed 2021-03-11 by kschiffer