TheThingsNetwork/lorawan-stack #3709: File Descriptor leak

virtualguy, 2021-01-28:
#### Summary
File Descriptor leak somewhere. Only occuring on clusters under a reasonable load

#### Steps to Reproduce
Run ttn-lw-stack for a few days with significant load

#### What do you see now?
[image]

Steady leak which results in a crash, most notably by failed redis store events.

#### What do you want to see instead?
No increase in open fds

#### Environment
$ttn-lw-stack version
The Things Stack for LoRaWAN: ttn-lw-stack
Version:             3.10.7
Build date:          2021-01-14T12:34:23Z
Git commit:          ecf52d6
Go version:          go1.15.6
OS/Arch:             linux/amd64

#### How do you propose to implement this?
.

#### How do you propose to test this?
Can deploy and monitor 'open fd' in our cluster

#### Can you do this yourself and submit a Pull Request?
Need help finding the leak, happy to provide further debugging information

KrishnaIyer, 2021-01-28:
Hey Jaime,

I'm gonna assign this for myself and re-assign if necessary.

Are you running TTS as a monolith? If not which component(s) is leaky?

> most notably by failed redis store events

Could you paste some logs with sensitive info retracted?

virtualguy, 2021-02-17:
Yes we are running as a monolith.

We have sent through a full set of logs from start to crash to TTI support (TTS-2378).

The rate of the leak increased substantially when we deployed additional gateways to the same site. I'm not sure if that's due to the increased duplicate uplink traffic or related to sending multicast (Class C) messages to clusters of gateways.

virtualguy, 2021-02-17:
Fyi, the number of open file descriptors is still increasing on 3.10.10 at the same rate

KrishnaIyer, 2021-02-18:
Thanks for the logs @virtualguy: I've reassigned this.

virtualguy, 2021-02-22:
I'll close this in favour of #3799 as its a duplicate

closed 2021-02-22 by virtualguy