TheThingsNetwork/lorawan-stack #3718: [console] Add Word Wrap option for messages in Live Data views

lnlp, 2021-01-29:
#### Summary
Add a Word Wrap option for messages in Live Data views. This is only needed for the Data Preview field.

#### Why do we need this?
To make it much simpler to read data from messages in Live Data views.
Messages are displayed as a single line but the lines are very long and the right part is not visible.
One has to horizontally scroll a message with a very tiny scrollbar below the message. In some situations this is not a problem, but having to do this all the time is a pain in the butt, especially when many messages are arriving and the list scrolls vertically automatically.

#### What is already there? What do you see now?
Long lines that fall off screen.

#### What is missing? What do you want to see?
Missing is a Word Wrap option for the Data Preview field.

#### How do you propose to implement this?
Add a checkbox somewhere in the top of Live Data windows to enable/disable the Word Wrap option.

kschiffer, 2021-02-16:
Blocked on #3817

KrishnaIyer, 2024-10-14:
Thanks for the issue and the long wait. This is a nice to have that we can do.
[image]

This is still relevant after the console redesign.

closed 2024-12-02 by KrishnaIyer