TheThingsNetwork/lorawan-stack #3719: [console] Scrolling with PgUp and PgDn keys does not scroll messages in Live Data views

lnlp, 2021-01-29:
#### Summary
Scrolling with PgUp and PgDn keys does not scroll messages in Live Data views.
Only works with mouse, no keyboard support.

#### Steps to Reproduce
In the Console open a Live Data window for gateway, application or device.
Wait until there is a long list of messages that don't fit on screen (e.g. a list three times the height of the window).
Activate the message list by clicking on it. Now try to scroll the messages with PgUp and PgDn keys. It just doesn't work. Instead this scrolls only the page not just the list of messages, which is useless.

#### What do you see now?
See Steps to reproduce.

#### What do you want to see instead?
Be able to scoll the list of messages with PgUp and PgDn keys.
Scrolling messages 'per page'.

#### How do you propose to implement this?
Speaks for itself.

htdvisser, 2021-02-01:
I don't think this is a bug, just missing functionality. @kschiffer please add bug label if you think otherwise.

kschiffer, 2021-02-01:
Indeed this is a feature request. @asmulko can you take a look at this?

kschiffer, 2021-02-09:
We will be working on focusing and cycling through individual event rows with the keyboard, as well as toggling their details using space/enter. Also PageUp and PageDown should jump to the first/last event.

Would that satisfy your needs @lnlp ?

lnlp, 2021-02-16:
> Would that satisfy your needs @lnlp ?

Yes I think so.

kschiffer, 2021-02-16:
Note for ourselves: this should also include the possibility to close the event details using the escape key #3717