TheThingsNetwork/lorawan-stack #3721: [bug][console] Unable to change Rx1 Delay after device has been created, option is missing

lnlp, 2021-01-31:
#### Summary
In [The Things Network V3 Console](https://eu1.cloud.thethings.network/console/):
Unable to change Rx1 Delay after device has been created.
This can now only be set when initially creating the device.
This will be a big problem for downlink messages via packet broker.
This is a problem for The Things Network V3 migration. Default value for Rx1 is too low for packet broker.
If user forgets to increase the value this cannot be changed afterwards and downlinks will not work.

#### Steps to Reproduce
Create new device and _after_ created try to find and option in the console to change Rx1 delay.

#### What do you see now?
Nada. Option to change Rx1 delay is missing for existing devices.

#### What do you want to see instead?
An option for changing the Rx1 delay, probably in the device's General settings.

#### Environment
[The Things Network V3 Console](https://eu1.cloud.thethings.network/console/).

#### How do you propose to implement this?
Add option to change Rx1 delay.

htdvisser, 2021-02-01:
This may need some expertise from @rvolosatovs about which fields to set exactly.

rvolosatovs, 2021-02-01:
Duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/2047 ?

bafonins, 2021-02-05:
@lnlp It is currently possible to change RX1 Delay via the Console.

1. Navigate to the general settings of you end device (`/console/applications/{application-id}/devices/{device-id}/general-settings`)
2. Expand `Network layer` section
3. Expand `Advanced settings`
4. Find `RX1 Delay` field

[image]

bafonins, 2021-02-05:
Closing this for now. @lnlp feel free to reopen if you still have issues with updating rx1_delay for your end device.

closed 2021-02-05 by bafonins

pomplesiegel, 2021-04-06:
Is this option only available for ABP devices? I do not see this option for any of my OTAA devices. thank you!