TheThingsNetwork/lorawan-stack #3724: TTNv2 / TTNv3 feature gap : Show frame counters (up/down) in TTN console device page

meten-natuurlijk, 2021-02-01:
#### Summary
The current TTNv2 console nicely shows the frame counters of a device in the Device page.
The TTNv3 console does not show this device frame-counter (yet).
This is useful for application/device maintainers, and needed to debug problems that are related to frame-counters.
For example : The new TTNv3 stack implements the LoRaWAN spec in a more strict way, which causes messages to be dropped if frame-counter are not within the expected range. 

#### Steps to Reproduce
The TTNv2 console shows the frame counters.
The TTNv3 console does not show the device frame-conuter (yet).

1. https://console.thethingsnetwork.org/applications/smal/devices/gps-27 
2. https://eu1.cloud.thethings.network/console/applications/smal/devices/gps-53

#### What do you see now?
Frame counters (up/down) are nicely displayed per device in the TTNv2 console

#### What do you want to see instead?
Frame counters (up/down) in the TTNv3 console

#### Environment
This problem is not related to a specific device or browser.

#### How do you propose to implement this?
Add the the frame counters to the Device page of the TTNv3 console.

#### How do you propose to test this?
* Program a (new) device
* Create a (new) application in the TTNv3 console
* Create a new device in the TTNv3 console
* Power-up the device, and watch the frame counter go up in the TTNv3 console
* Power-down the device and wait 30 minutes
* Make sure that the last frame counter stays visible on the Device page of the TTNv3 console

#### Can you do this yourself and submit a Pull Request?
No, sorry, I don't have the software development-skills for this.

I think this issue is related to issue #2230

htdvisser, 2021-02-02:
Also related to https://github.com/TheThingsNetwork/lorawan-stack/issues/2434

kschiffer, 2021-02-02:
The frame counters should appear once a first uplink/downlink event arrives. They aren't shown initially after a refresh, that functionality was for some reason removed [here](https://github.com/TheThingsNetwork/lorawan-stack/commit/6ebbf547e369ac4ae9faffe7b14718408bf6a650#diff-a88642f3f676f38b8705092cdb9f2247e9021a2c237548e418e16c67b2edd278), which I think (after consulting with @rvolosatovs) was by mistake (cc @bafonins).

I'll file a PR to re-add it so the frame counters will be shown initially as well.

bafonins, 2021-02-02:
Please see https://github.com/TheThingsNetwork/lorawan-stack/issues/3230 for more details

kschiffer, 2021-02-02:
Appears like the confusion originated from:

> Console should **always and only ever** use `full_f_cnt` field on uplinks, which is set by Network Server. It **must** discard all other uplink FCnt values.

cc @rvolosatovs

rvolosatovs, 2021-02-02:
> Appears like the confusion originated from:
> 
> > Console should **always and only ever** use `full_f_cnt` field on uplinks, which is set by Network Server. It **must** discard all other uplink FCnt values.
> 
> cc @rvolosatovs

What I meant here is that of the fields present in the uplink message itself, only `full_f_cnt` is to be used to determine FCnt of said uplink. `session.last_f_cnt_up` still represents the FCntUp of the session and is most likely equal to the `full_f_cnt` of last uplink.

PR #3731 (2021-02-09): Fix initial display of end device frame counts
#### Summary
Closes #3724 

#### Changes
- Apply frame counts of the EDs NS session on the initial fetch
- Always show frame counter icons in the ED title section (but use `n/a`, if not available)

#### Testing

- Manual (will add e2e once we can to traffic simulation)

##### Regressions

None.

#### Notes for Reviewers
I now use the network downlink frame counter. Not entirely sure whether that is the preferred value to display in the Console.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] C [... cut]
closed 2021-02-09 by kschiffer