TheThingsNetwork/lorawan-stack #3729: Create user API Key without a browser login

KrishnaIyer, 2021-02-02:
#### Summary

Create user API Key without a browser login

#### Why do we need this?

For seeding during automated testing

#### What is already there? What do you see now?

`is-db create-admin-user` creates an admin user.

#### What is missing? What do you want to see?

To create a user API key to be used with CLI scripts, the user needs to login (with or without callback) and then create an API key which can be used for further calls from the CLI. This does not work for full automation without a browser.
- Admin users can create API Keys for other users via the CLI but the admin user itself needs a browser for initial login.

#### Environment

`tts v3.10.9`

#### How do you propose to implement this?

Add an option to create a user API key to `is-db create-admin-user`

#### How do you propose to test this?

Update `tools/bin/mage dev:initStack` and test that API Keys get created and that those can be used via the CLI without a login.

#### Can you do this yourself and submit a Pull Request?

I think so. @htdvisser  please advice if this is alright and/or if there's an easier way to achieve this.

htdvisser, 2021-02-02:
Duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/3518

Implemented in https://github.com/TheThingsNetwork/lorawan-stack/pull/3622

closed 2021-02-02 by htdvisser

KrishnaIyer, 2021-02-02:
Fantastic!