TheThingsNetwork/lorawan-stack #3736: Can't run ttn-lw-stack commands after building with goreleaser

kurtmc, 2021-02-03:
#### Summary

I cannot run any `ttn-lw-stack` commands using the binaries produced by running goreleaser.

#### Steps to Reproduce

1. Checkout tag v3.10.7
2. Run go releaser
```
go run github.com/goreleaser/goreleaser --snapshot -f .goreleaser.snapshot.yml --rm-dist
```
3. Try a command
```
./dist/stack_linux_amd64/ttn-lw-stack start
```

#### What do you see now?

This is the error produced:

```
panic: reflect: call of reflect.Value.Type on zero Value

goroutine 1 [running]:
reflect.Value.Type(0x0, 0x0, 0x0, 0x25bb290, 0x2081d60)
	/usr/local/go/src/reflect/value.go:1879 +0x183
github.com/mitchellh/mapstructure.DecodeHookExec(0x216ff20, 0x25bb290, 0x0, 0x0, 0x0, 0x221a2e0, 0xc000403d88, 0x196, 0x0, 0x0, ...)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/decode_hooks.go:46 +0x1cb
github.com/mitchellh/mapstructure.ComposeDecodeHookFunc.func1(0x2081d60, 0x40704e0, 0x98, 0x221a2e0, 0xc000403d88, 0x196, 0x0, 0xabc1757d01551b00, 0xc000ce7b90, 0xc000b4f968)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/decode_hooks.go:67 +0xd1
github.com/mitchellh/mapstructure.DecodeHookExec(0x2147b80, 0xc00049ec60, 0x2081d60, 0x40704e0, 0x98, 0x221a2e0, 0xc000403d88, 0x196, 0x16, 0x2e, ...)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/decode_hooks.go:50 +0x12b
github.com/mitchellh/mapstructure.(*Decoder).decode(0xc000587d98, 0xc000ce7b90, 0x2e, 0x2081d60, 0x40704e0, 0x221a2e0, 0xc000403d88, 0x196, 0x2e, 0x0)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:416 +0x8b1
github.com/mitchellh/mapstructure.(*Decoder).decodeStructFromMap(0xc000587d98, 0xc0013dde00, 0x17, 0x2157840, 0xc000d08e70, 0x15, 0x23db400, 0xc000403d78, 0x199, 0x0, ...)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:1357 +0x1be4
github.com/mitchellh/mapstructure.(*Decoder).decodeStruct(0xc000587d98, 0xc0013dde00, 0x17, 0x2157840, 0xc000d08e70, 0x23db400, 0xc000403d78, 0x199, 0x2157840, 0xc000d08e70)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:1183 +0x621
github.com/mitchellh/mapstructure.(*Decoder).decode(0xc000587d98, 0xc0013dde00, 0x17, 0x2157840, 0xc000d08e70, 0x23db400, 0xc000403d78, 0x199, 0x17, 0x0)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:439 +0x765
github.com/mitchellh/mapstructure.(*Decoder).decodeStructFromMap(0xc000587d98, 0x1f6449d, 0x2, 0x2157840, 0xc000c909f0, 0x15, 0x241a960, 0xc000403cd0, 0x199, 0x0, ...)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:1357 +0x1be4
github.com/mitchellh/mapstructure.(*Decoder).decodeStruct(0xc000587d98, 0x1f6449d, 0x2, 0x2157840, 0xc000c909f0, 0x241a960, 0xc000403cd0, 0x199, 0x2157840, 0xc000c909f0)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:1183 +0x621
github.com/mitchellh/mapstructure.(*Decoder).decode(0xc000587d98, 0x1f6449d, 0x2, 0x2157840, 0xc000c909f0, 0x241a960, 0xc000403cd0, 0x199, 0x94, 0x0)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:439 +0x765
github.com/mitchellh/mapstructure.(*Decoder).decodeStructFromMap(0xc000587d98, 0x0, 0x0, 0x2157840, 0xc00099cc00, 0x15, 0x241cd20, 0xc000403000, 0x199, 0x0, ...)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:1357 +0x1be4
github.com/mitchellh/mapstructure.(*Decoder).decodeStruct(0xc000587d98, 0x0, 0x0, 0x2157840, 0xc00099cc00, 0x241cd20, 0xc000403000, 0x199, 0x2157840, 0xc00099cc00)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:1183 +0x621
github.com/mitchellh/mapstructure.(*Decoder).decode(0xc000587d98, 0x0, 0x0, 0x2157840, 0xc00099cc00, 0x241cd20, 0xc000403000, 0x199, 0xc0002cb300, 0xc00099cc00)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:439 +0x765
github.com/mitchellh/mapstructure.(*Decoder).Decode(0xc000587d98, 0x2157840, 0xc00099cc00, 0x0, 0xc00049ec60)
	/go/pkg/mod/github.com/mitchellh/mapstructure@v1.4.0/mapstructure.go:374 +0xf0
go.thethings.network/lorawan-stack/v3/pkg/config.(*Manager).Unmarshal(0xc0008f2c00, 0x2188d80, 0xc000403000, 0xc0006cbc00, 0xc000dcfab0)
	/tmp/build/80754af9/lorawan-stack/pkg/config/config.go:235 +0x2bf
go.thethings.network/lorawan-stack/v3/cmd/ttn-lw-stack/commands.glob..func7(0x3fe9e00, 0x406e568, 0x0, 0x0, 0x0, 0x0)
	/tmp/build/80754af9/lorawan-stack/cmd/ttn-lw-stack/commands/root.go:60 +0x7d
github.com/spf13/cobra.(*Command).execute(0x3fe9e00, 0x406e568, 0x0, 0x0, 0x3fe9e00, 0x406e568)
	/go/pkg/mod/github.com/spf13/cobra@v1.1.1/command.go:829 +0x55e
github.com/spf13/cobra.(*Command).ExecuteC(0x3fe9b60, 0x0, 0x2068420, 0xc0000540b8)
	/go/pkg/mod/github.com/spf13/cobra@v1.1.1/command.go:958 +0x349
github.com/spf13/cobra.(*Command).Execute(...)
	/go/pkg/mod/github.com/spf13/cobra@v1.1.1/command.go:895
main.main()
	/tmp/build/80754af9/lorawan-stack/cmd/ttn-lw-stack/main.go:26 +0x2d
```

#### What do you want to see instead?

[... 48 lines cut ...]
	/go/pkg/mod/github.com/spf13/cobra@v1.1.1/command.go:895
main.main()
	/tmp/build/80754af9/lorawan-stack/cmd/ttn-lw-stack/main.go:26 +0x2d
```
```
# ./dist/cli_linux_amd64/ttn-lw-cli version
The Things Network Command-line Interface: ttn-lw-cli
Version:             v3.10.7-SNAPSHOT-ecf52d6b5
Build date:          2021-02-03T01:44:28Z
Git commit:          ecf52d6b5
Go version:          go1.15.7
OS/Arch:             linux/amd64
```

#### How do you propose to implement this?

N/A

#### How do you propose to test this?

N/A

#### Can you do this yourself and submit a Pull Request?

No.

rvolosatovs, 2021-02-03:
Thanks for report - it does not seem like an issue specific to goreleaser though and is most probably caused by your environment and bug in handling of it.
Which configuration options do you have set in your config/environment?

virtualguy, 2021-02-03:
@rvolosatovs the logs reported are with a clean env and no config specified

kurtmc, 2021-02-03:
Yeah it looks related to goreleaser, I switched our tooling to just run `go build` and it seems to have resolved the issue.

htdvisser, 2021-02-04:
This stack trace sounds to me like it's a bug in `github.com/mitchellh/mapstructure@v1.4.0`, which is also interesting, because we depend on `github.com/mitchellh/mapstructure@v1.3.0`.

htdvisser, 2021-02-04:
It looks like this is https://github.com/mitchellh/mapstructure/issues/231 which was fixed in https://github.com/mitchellh/mapstructure/pull/232 and released in https://github.com/mitchellh/mapstructure/releases/tag/v1.4.1.

kurtmc, 2021-02-04:
@htdvisser Here is a one liner that you can use to reproduce the issue if that helps:
```
docker run -it golang:1.15 bash -c "git clone --depth 1 --branch v3.10.7 https://github.com/TheThingsNetwork/lorawan-stack.git; cd lorawan-stack; make init; go run github.com/goreleaser/goreleaser --snapshot -f .goreleaser.snapshot.yml --rm-dist; ./dist/stack_linux_amd64/ttn-lw-stack start"
```

htdvisser, 2021-02-05:
Ah, thanks, this is helpful.

Here's what seems to happen:

- GoReleaser is not in our main `go.mod` (but in `tools/go.mod`)
- If you execute `go run github.com/goreleaser/goreleaser`, Go downloads the latest version of GoReleaser, and updates our main `go.mod`
	- You can see this when running `git diff` afterwards
- Because the latest GoReleaser version (currently) depends on `github.com/mitchellh/mapstructure v1.4.0`, it also upgrades that dependency for The Things Stack.
- When you build The Things Stack, it will use that `v1.4.0` version, instead of the `v1.3.0` version that we know worked for us.

It looks like we need to update our documentation a bit. In the meantime, you can use this:

```
git clone --depth 1 --branch v3.10.7 https://github.com/TheThingsNetwork/lorawan-stack.git
pushd lorawan-stack
make init
pushd tools
go build -o bin/goreleaser github.com/goreleaser/goreleaser
popd
tools/bin/goreleaser --snapshot -f .goreleaser.snapshot.yml --rm-dist
./dist/stack_linux_amd64/ttn-lw-stack start
```

@rvolosatovs is this correct or is there a better way?

rvolosatovs, 2021-02-08:
I'm afraid not at the moment

closed 2021-02-25 by htdvisser

htdvisser, 2021-05-04:
When upgrading mapstructure to v1.4.1 (in https://github.com/TheThingsNetwork/lorawan-stack/pull/4124) this error still happens, so it may actually be something else...

PR #3821 (2021-02-25): Fix documentation of GoReleaser snapshots

#### Summary

This pull request fixes the instructions for building (snapshot) releases with GoReleaser.

Closes #3736

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [x] Documentation: Relevant documentation is added or updated.
- [ ] Changelog: Significant features, behavior changes, deprecations and fixes [... cut]