TheThingsNetwork/lorawan-stack #3741: Webhook doesn't work for "Downlink sent" option

TheIronNinja, 2021-02-04:
#### Summary
Webhook doesn't work for "Downlink sent" option

#### Steps to Reproduce
Connect the application to a controled webhook so you can see which packets are sent and which not.
Using an existing device and gateway, simulate or generate a downlink.

#### What do you see now?
The "Downlink Scheduled" packet is sent, but once it's sent the webhook isn't triggered. I can see that the packet is sent from the Console view of the device and from the live view of the Gateway, so I know the packet is being sent and received.

#### What do you want to see instead?
The webhook being called whenever a Downlink is sent from the gateway.

#### Environment
The Things Stack v3.10.0 (I've checked the changelog and this hasn't been addressed yet so it should still be relevant)

KrishnaIyer, 2021-02-05:
@neoaggelos : Do we already support sending TX acks to the application server?

adriansmares, 2021-02-05:
This needs https://github.com/TheThingsNetwork/lorawan-stack/issues/56, which is blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/76 due to priorities.

neoaggelos, 2021-06-24:
Closed with https://github.com/TheThingsNetwork/lorawan-stack/pull/4282.

closed 2021-06-24 by neoaggelos