TheThingsNetwork/lorawan-stack #3754: Downlink stop working for devices after a period

cnoork, 2021-02-09:
#### Summary
I have installed two mclimate vicki thermostats and control them by downlinks. They send now every 4 minutes uplinks and the control from the valve is done by downlinks. Only after one day the downlinks don't work anymore. In the console I get the message: "error:pkg/networkserver:unknown_session (unknown session)"

Initial the devices sends every 3 minutes a uplink but changed this to 4 minutes in the hope this was the issue.

Because the name of the device need to be changed I deleted the device and recreated it, this made the downlink work again. The next time it happend I restarted the device and with the rejoin it did work also again.

#### Steps to Reproduce
Till now I noticed that it happens after a time but do not know if time or number of messages is relevant. But sending frequent downloads it will stop. Rejoin the device will solve it for a small period (do not know how many hours).

#### What do you see now?
With mqtt I do not see a different msg, on the console I get "error:pkg/networkserver:unknown_session (unknown session)"

#### What do you want to see instead?
successful downlinks

#### Environment
TTN --> v3.11.0-rc1

#### How do you propose to implement this?
I think this is an issue in the back-end

#### How do you propose to test this?
Send many downlinks

#### Can you do this yourself and submit a Pull Request?
No, unfortunately

[image]

rvolosatovs, 2021-02-11:
Seems that AS is using the wrong session when replacing downlinks

rvolosatovs, 2021-03-03:
@adriansmares I guess this is going to be closed by #3649 ?

adriansmares, 2021-03-03:
> @adriansmares I guess this is going to be closed by #3880 ?

Yes.

closed 2021-03-03 by adriansmares