TheThingsNetwork/lorawan-stack #3755: Add copy&paste tools to the Session Key fields of a device

tiefpunkt, 2021-02-09:
#### Summary

In the EUI and RootKey fields of a device you can alter the way the keys are displayed for easier copy&paste into applications, like use the array representation instead of just the string of bytes. This function is missing for the Session Keys. For ABP devices this would be great to have still (it was available in the v2 UI)

#### Why do we need this?

Ease of use, similar user experience to the old UI

#### What is already there? What do you see now?

Right now you can only copy the byte string.
[image]

#### What is missing? What do you want to see?

Same UI representation as for the Root Keys.
[image]

#### Environment

eu1 public installation

#### How do you propose to implement this?

From what I can see, this should be a fairly simple change, changing https://github.com/TheThingsNetwork/lorawan-stack/blob/77f9c699f5f0b1b017c7d11cee377ab8cdd26615/pkg/webui/console/views/device-overview/index.js#L185 from "code" to "byte", and the sections for the othter fields below that.

#### How do you propose to test this?

Check in the UI

#### Can you do this yourself and submit a Pull Request?

I surely could, but I don't have the means to test it right now, so I'd like to get some feedback in on this first.

tiefpunkt, 2021-02-09:
Looks like @kschiffer beat me to it with #3752 🙈

closed 2021-02-09 by tiefpunkt

kschiffer, 2021-02-09:
Hahah, sorry about that :). Still thank you for taking the time to write this descriptive issue!