TheThingsNetwork/lorawan-stack #3771: Configure routing policies

johanstokking, 2021-02-09:
#### Summary

Configure routing policies.

For listing available networks and tenants, we need https://github.com/packetbroker/iam/issues/2.

#### Why do we need this?

For networks to configure routing policies with other networks

#### What is already there? What do you see now?

This should currently be done with [`pbctl`](https://github.com/packetbroker/pb#configure-routing-policies), but it would be nice to be able to configure routing policies via the CLI and Console.

#### What is missing? What do you want to see?

The ability to configure routing policies between the local Packet Broker network or tenant, and any other network or tenant.

#### How do you propose to implement this?

- Add API to Packet Broker Agent for routing policy management
- Implement CLI support
- Implement Console support. Let's start with manually entering the NetID and tenant ID. With https://github.com/packetbroker/iam/issues/2, we can have a dropdown of public networks and tenants (but still manual entry must be possible)

#### How do you propose to test this?

Integration testing

#### Can you do this yourself and submit a Pull Request?

Can do API and CLI support.

@kschiffer please already think about UX. See linked `pbctl` commands to get an idea of what can be done.

johanstokking, 2021-03-12:
@kschiffer please start thinking about the UX in the form of wireframes.

The API is designed. See https://github.com/TheThingsNetwork/lorawan-stack/pull/3912/files#diff-4371dadc483aa14e23af85a7baf0e1d57b6d0c67fcf203b94412e13b224cc541 and then especially those `rpc`s.

My idea is as follows:

- Administrators can manage peering for their network. In The Things Stack, this is per deployment. In The Things Stack Cloud, this is per tenant. That is, no application or device level settings (just yet)
- The configuration is technically two-fold: forwarder and home network configuration. Forwarders have gateway infrastructure, home networks have end devices. Most users will play either role, but technically we need to make a distinction here

The UI elements are:

1. Top-level menu item for Packet Broker. My suggestion would be to use the term Packet Broker to push that brand. So the menu item would read **Packet Broker** and not peering
2. In the Console overview page, show the status of Packet Broker (t.b.d., but for now whether the network has an active registration)
3. The Packet Broker settings page. Here, the user can:
   - Register and deregister with Packet Broker, and see their registration info (`/pba/registration` bindings). When there is no registration (`NotFound`), disable all components on the page
   - Manage the forwarder routing policies. This is on two levels: the default settings (`/pba/forwarder/default-policy` bindings) and per-home network (`/pba/forwarder/policies` bindings) settings. To configure per home network, you can allow the user to select a home network from a list of all available networks (`/pba/forwarde/home-networks` binding)
   - View the routing policies that other forwarders set for this home network (`/pba/home-network/policies` binding)

There will be statistics in the future, but don't bother with that for now.

johanstokking, 2021-03-14:
This needs major API extensions so this is moved to 3.12.

johanstokking, 2021-03-30:
@kschiffer please coordinate UX implementation and triage

kschiffer, 2021-04-13:
See the wireframes below:

[image]

[210413_packet_broker_ux.pdf](https://github.com/TheThingsNetwork/lorawan-stack/files/6302208/210413_packet_broker_ux.pdf)

johanstokking, 2021-04-13:
Looks and works great this way.

Looking at the API calls, it seems like:

- List Forwarder policies: `/pba/forwarders/policies` (load all pages)
- For the **All networks** tab: `/pba/home-networks` (load current page)
   - For **Their routing policy towards us**: lookup the Forwarder policy [1]
   - For **Our routing policy for them**: `/pba/home-networks/policies/{net_id}/{tenant_id}`
- For the **Networks with non-default policies** tab: what is non-default here; their and/or our routing policy? That indeed requires a filter, like: `/pba/home-networks?require_specific_policy=true`
- Filtering: full text search, add `name_contains` and `id_contains`

Anything else?

kschiffer, 2021-04-13:
> *   For the **Networks with non-default policies** tab: what is non-default here; their and/or our routing policy? That indeed requires a filter, like: `/pba/home-networks?require_specific_policy=true`

Yeah I'd say both. Should be the best bet for policies that users are specifically interested in, so we can use that until we fleshed out network discovery.

> Filtering: full text search, add `name_contains` and `id_contains`

Just to be sure, this needs to be OR, not AND when both are in use.

> Instead of \[2\], we can also prefetch all Home Network policies. TBH, I think this is cleaner. Otherwise you get 2\*N API calls (with N the number of rows).

Agreed. It's good to have an RPC for individual policies but it should (also) be possible to fetch all policies at once.

johanstokking, 2021-04-13:
OK, so you prefetch all Forwarder and Home Network routing policies. That is already possible.

There are two API additions then:

1. A parameter for `/pba/home-networks` to list only Home Networks with whom a specific routing policy is defined (in either direction)
2. A full text search on name and tenant ID.

---

> > Filtering: full text search, add `name_contains` and `id_contains`
> 
> Just to be sure, this needs to be OR, not AND when both are in use.

Really?

kschiffer, 2021-04-13:
> Really?

Using the filter on the table should match both ID and name.

johanstokking, 2021-04-13:
> > Really?
> 
> Using the filter on the table should match both ID and name.

Ah I see what you mean now. Yes, indeed.

johanstokking, 2021-04-13:
I will pick up the API part.

johanstokking, 2021-06-24:
Closed by #4281

PR #3982 (2021-03-29): Configure Packet Broker registration and routing policies

#### Summary

Closes #3771 
Implements https://github.com/packetbroker/api/issues/23
Implements https://github.com/packetbroker/api/issues/25
References #2798 by removing and avoiding `gogoproto` options
Documentation filed in https://github.com/TheThingsIndustries/lorawan-stack-docs/pull/284

#### Changes

- Add `ttn-lw-cli packetbroker` commands to manage the registration and routing policies
- Add Packet Broker Agent configuration and `Pba` service to facilitate that
- While we're at it: include the gateway ID and map our `ISM_2400` [... cut]
closed 2021-06-24 by johanstokking