TheThingsNetwork/lorawan-stack #3776: Server on ttn.fyi reports 502 Bad Gateway

sollie, 2021-02-10:
#### Summary
Webserver on ttn.fyi has been unavailable for some time, making the Getting Started guide and getting invites to Slack unavailable.

...

#### Why do we need this ?
Resources helps new users getting started and allows people to join the Slack.
...

#### What is already there? What do you see now?
N/A
...

#### What is missing? What do you want to see?
N/A
...

#### How do you propose to document this?
N/A
...

#### Can you do this yourself and submit a Pull Request?
No
...

johanstokking, 2021-02-11:
Thanks @sollie for reporting. We're looking into it.

This is not the right place for this issue, but I filed it internally.

closed 2021-02-11 by johanstokking

johanstokking, 2021-02-12:
@sollie can you check if this is now fixed for you?

sollie, 2021-02-12:
@johanstokking Looks good now, thanks.

I also agree this was probably not the correct place for this issue, but it looked like the category that was closest when I opened it.