TheThingsNetwork/lorawan-stack #3777: Inconsistent usage of gateway ID in global_conf.json

nejraselimovic, 2021-02-10:
#### Summary
Found this on our forum: https://www.thethingsnetwork.org/forum/t/v3-download-global-conf-json/43973/3

[image]

#### What do you see now?
Gateway EUI from the Console is stored as `gateway_ID` in `global_conf.json` file that's downloadable from the Console. Gateway ID in the Console is the mnemonic name of the gateway.

#### What do you want to see instead?
Can this be reworded somehow? It's simple, but I guess it can create a confusion.

#### Can you do this yourself and submit a Pull Request?
No. @kschiffer I assigned you, please reassign if it's not your area

kschiffer, 2021-02-12:
Maybe @rvolosatovs can make a comment on this. I suppose though that we cannot change the structure of the `global_conf.json`. Changing the name of the gateway ID would create only more confusion I'd say. So I guess our best bet is to clarify this in the docs. If there's a doc page for the gateway conf already, we can link it there with a help icon.

nejraselimovic, 2021-02-12:
> Maybe @rvolosatovs can make a comment on this. I suppose though that we cannot change the structure of the `global_conf.json`. Changing the name of the gateway ID would create only more confusion [...]

I also saw Krishna's response on the forum after, it's some legacy stuff, so I guess I'll just take this one and clarify it in the documentation.

KrishnaIyer, 2021-02-15:
Indeed I responded to a similar query on the forum; https://www.thethingsnetwork.org/forum/t/v3-download-global-conf-json/43973/6

closed 2021-03-31 by nejraselimovic