TheThingsNetwork/lorawan-stack #3785: Console lacks context sensitive help

lnlp, 2021-02-11:
#### Summary
The console lacks context sensitive help.
Add consistent context sensitive help.

#### Why do we need this?
Many fields in the console (e.g. during adding/configuring Gateways, Applications and Devices) lack any form of help leaving the user puzzled and unclear whether (s)he should fill in the field on not, what it's for and what possible values are.
For some fields this is obvious but for many other fields it is not (e.g. add device | advanced network settings | factory preset frequencies but this is only one exmple).

The small descriptive sentence below the field is often not sufficient (at all).
The forms should contain _context sensitive_ help. For entering a DevEUI one can click a question mark button which then shows "See DEVEUI in the Glossary". That is not context sensitive help but a 'textual pointer', not a link and provides poor UX.
Many other fields do not even have that option and all that resides is a small link bottom right on the page to the main The Things Stack Documentation site root. That is not very helpful at all and often frustrating.
Results in poor UX that needs major improvement.

#### What is already there? What do you see now?
See above. All form fields should be checked.

#### What is missing? What do you want to see?
Context sensitive help.
Popup a larger description with help about the field _and_ include a link where the subject is fully explained in the documentation.

#### Environment
V3 Console in general.

#### Can you do this yourself and submit a Pull Request?
No.

lnlp, 2021-02-11:
See:

https://github.com/TheThingsNetwork/lorawan-stack/issues/2813#issuecomment-652635759

> This is going to change drastically once we have the device repository in place. It will be a matter of selecting the vendor, the model, and potentially the region and (if applicable) the firmware version. That's it.
> 
> So even though we should make things easy to input, the best thing is to avoid users to input this stuff in the first place.

That's not "That's it" for TTN V3 with a community with many DIY devices.

This will not work for DIY devices. And 'devices' (manual settings) is one of the console area that needs context sensitive help.

kschiffer, 2021-02-11:
Thanks for filing this issue @lnlp.

We are aware of this issue and already have tooltips with contextual help in our pipeline and their content is also already been provided by @benolayinka. I'll make sure that the priorities are updated to have this issue resolved timely.

johanstokking, 2021-02-12:
Let's make sure we explain the "factory preset frequencies" too; and probably all the other ABP settings.

johanstokking, 2021-02-12:
Rescheduling to March. Let's start with the low hanging fruit.

kschiffer, 2021-02-15:
Superseded by #3809.

@lnlp please follow this issue to track our progress on this.

closed 2021-02-15 by kschiffer