TheThingsNetwork/lorawan-stack #3791: Basic Station websockets not on port 443

jpmeijers, 2021-02-12:
#### Summary

The main reason for choosing websockets for Basic Station, above say mqtt or grpc, was that websockets run on port 80 and 443. Websockets also look like normal http traffic when they are set up. The reason this is good is because many company firewalls block all unknown ports, and normally will allow at least port 80 and 443 for web traffic. Making the gateway's traffic look like web traffic makes using gateways on company networks much easier.

Today some community members on slack ran into the issue that their company network blocks Basic Station. After investigation we saw that TTSv3 uses port 8887 for Basic Station (https://thethingsindustries.com/docs/reference/networking/), not the expected 443 like on v2 (https://www.thethingsnetwork.org/docs/gateways/basics-station/index.html).

#### Steps to Reproduce

...

#### What do you see now?

Basic Station gateways need to use port 8887.

#### What do you want to see instead?

Basic Station gateways should use port 443.

#### Environment

TTS v3.11

#### How do you propose to implement this?

If we can't piggyback on the existing web port 443, maybe run a proxy service next to TTNv3.

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

Nope

jpmeijers, 2021-02-12:
Andy:de:  34 minutes ago
Hi,
as I'm running a gateway behind a corporate firewall (for devlopment), other ports than 80 and 443 are blocked.
The TTN v3 Stack uses the (TLS) port 8887 for BasicStation and therefore no connection can be established.
Do you have any idea how to establish the communication over port 443 or to set an proxy (e.g. wstunnel) but still use the TLS protocol?

jpmeijers  18 minutes ago
Are you sure about this? I though the main reason for basic station was to use the standard https port to get past firewalls

rroemhild  17 minutes ago
https://thethingsindustries.com/docs/reference/networking/

jpmeijers  17 minutes ago
https://www.thethingsnetwork.org/docs/gateways/basics-station/index.html
Server Address: wss://lns.{eu|us|in|au}.thethings.network:443

rroemhild  16 minutes ago
That is for TTNv2

jpmeijers  16 minutes ago
Oh damn, so they broke the whole point behind BS in V3
:man-shrugging:
1

jpmeijers  15 minutes ago
@Andy you could just use V2 for now, as the data is anyway forwarded to V3. And maybe file an issue on github about this, as I believe this is a bug

htdvisser  10 minutes ago
I'm afraid this is not a bug

jpmeijers  10 minutes ago
But what's the point of BasicStation then?

mlo  10 minutes ago
stunnel to the rescue?

htdvisser  8 minutes ago
We've seen some "sub-optimal" TLS implementations on several Basic Station gateways that prevent us from serving it on the same port as normal web traffic without making compromises that we don't want to make

mlo  7 minutes ago
I guess they could only handle TLS < 1.1 or similar?

jpmeijers  7 minutes ago
For private instances sure, but can't we do the same as for V2, where we have a separate hostname to which basic stations gateways can connect on port 443?

jpmeijers  6 minutes ago
I'll create a github issue for this anyway. Then it's discussed centrally in a better place.

htdvisser  6 minutes ago
I don't remember the exact details, but there were indeed some interesting shortcuts to make everything work with limited gateway resources
New

Andy:de:  2 minutes ago
Thanks guys for the quick response!   I tried to migrate our development gateway to v3 stack and got stuck... back to v2 until our admins opened the port for me :wink:

KrishnaIyer, 2021-02-15:
Well this maybe be ok for many deployments. The main reason we chose different ports for TTSC is the use of different Root CA for the CUPS/LNS end points and a more commonly trusted (ex: LE) for the console (web-browser).

KrishnaIyer, 2021-04-30:
This is actually deployed and can be closed

closed 2021-04-30 by KrishnaIyer

AndreasAmMueller, 2021-05-01:
What's the correct uri now? wss://eu1.cloud.thethings.network or something else?

My gateway is currently not able to connect to that socket or is it a part of 3.12.2 (scheduled for may 3rd)