TheThingsNetwork/lorawan-stack #3795: Tenant ID and Cluster ID in uplink message json

jpmeijers, 2021-02-13:
#### Summary

A third party service might need to know which Network Server received and forwarded the uplink message. Currently the uplink json looks the same for different tenants on a cluster, as well as the same as uplink json data from other clusters.

The issue was also described in https://github.com/TheThingsNetwork/lorawan-stack/issues/3705.

A cluster is aware which tenant is forwarding the message, thus the cluster can add the tenant id into the json object. Most likely the cluster is also aware of its own id, because the cluster needs this to forward and identify data to the Packet Broker. In that case we can also add the cluster ID into the json object.

#### Why do we need this?

For 3rd party services to know from which LoRaWAN network server the uplink message was received.

#### What is already there? What do you see now?

One can identify the device and the application, but not the tenant and the cluster.

#### What is missing? What do you want to see?

I also want to identify the tenant and the cluster.

#### Environment

TTN v3.11.0-rc2

#### How do you propose to implement this?

Add the cluster ID and the tenant ID into the end device identifiers:

```
   "end_device_ids":{
      "device_id":"eui-0004a30b001cXXXX",
      "application_ids":{
         "application_id":"jpm-test"
      },
      "dev_eui":"0004A30B001CXXXX",
      "join_eui":"0000000000000000",
      "dev_addr":"260BXXXX",
      "tenant_id":"ttn",
      "cluster_id":"eu1.cloud.thethings.network" // or just "eu1"
   },
```

#### How do you propose to test this?

Check the uplink json format.

#### Can you do this yourself and submit a Pull Request?

No

johanstokking, 2021-02-17:
I believe we already provide the tenant ID via the context, but then base64 encoded.

For the originating cluster, how about a HTTP header?

I feel like this doesn't really belong in the payload itself.

htdvisser, 2021-02-17:
We could indeed add some `X-TTS-DOMAIN: tenant.eu1.cloud.provider.domain` header, but as I already [commented in the other issue](https://github.com/TheThingsNetwork/lorawan-stack/issues/3705#issuecomment-779063521):

> Much better would be if your users register the necessary parameters in your application, and then you give them some token (that identifies those parameters) and let them configure that in the webhook template, so that the token is included in every request.

I don't think external applications should rely only on some header or payload field to determine which network sent a webhook. A malicious actor can then easily send fake webhooks. You wouldn't have this issue if you use that token approach I suggested in the other issue.

So while we definitely can add a header, I'm afraid that doing so would make it too easy for developers to build insecure applications. By not doing it, and documenting some recommendations, we can help them build better applications.

jpmeijers, 2021-02-17:
Thinking about the end goal here again, which is to know to which network a gateway, reported in the metadata, belongs. So from my perspective the tenant and cluster should go together with the gateway id.

But doing this:
`X-TTS-DOMAIN: tenant.eu1.cloud.provider.domain`
would be excellent, as that provides the unique identification one needs to know to which network the gateways are connected.

But how do we solve this for MQTT? Not that I need it, but shouldn't the two integrations provide the same data?

The end goal here is not authorization. It's purely identifying the devices/gateways in the uplink. For authorization I agree that the best would be for end users to enter a token obtained from the external service. But there is no guarantee that end users will provide the correct network server instance when requesting a token. Also nothing stops them from using the same token on multiple networks (au1, eu1, nam1, private, etc). The token therefore does not solve the main issue here.

johanstokking, 2021-02-17:
> But how do we solve this for MQTT? Not that I need it, but shouldn't the two integrations provide the same data?

With MQTT you don't need it as you dial out to a specific address already.

> The end goal here is not authorization

I agree, security is not a concern. The header would be informative only. The receiving endpoint needs other means to verify authenticity and integrity anyway, regardless of whether we add this header or not.

htdvisser, 2021-02-17:
As long as documentation is clear about it being informative I'm fine with it. Adding this shouldn't be too much work, because it's very similar to what we already do with `createDownlinkURL`.

htdvisser, 2021-02-17:
I don't think there's much more to discuss, so let's work on this in March. The scope only includes adding an `x-tts-domain: tenant.eu1.cloud.provider.domain` header to "up" webhooks. Outgoing pub/sub integrations are out of scope for this issue.

This may need additional config, as I'm not sure we should use the `PublicAddress`/`PublicTLSAddress` in `DownlinksConfig` for this.

closed 2021-03-31 by adriansmares