TheThingsNetwork/lorawan-stack #3798: Change the footer link to thethingsindustries.com

mattrude, 2021-02-14:
#### Summary

The footer of each page points to https://www.thethingsnetwork.org/,  this implies that you are connected to the The Things Network.

The confusion comes in since a vanilla install of The Things Stack dose not connect back to The Things Network.

I propose to change the link from https://www.thethingsnetwork.org to https://www.thethingsindustries.com/ to better reflect the actual product/software the user is using.

#### Steps to Reproduce

Click on the link in the footer.

#### What do you see now?

A link to [thethingsnetwork.org](https://www.thethingsnetwork.org)

#### What do you want to see instead?

A link to [thethingsindustries.com](https://www.thethingsindustries.com)

#### Environment

```The Things Stack for LoRaWAN: ttn-lw-stack
Version:             3.11.0
Build date:          2021-02-10T13:20:50Z
Git commit:          cc9bbb42b
Go version:          go1.15.7
OS/Arch:             linux/arm
```

#### How do you propose to implement this?

```
diff --git a/pkg/webui/components/footer/index.js b/pkg/webui/components/footer/index.js
index 4fcd344bf..c0ac33d9d 100644
--- a/pkg/webui/components/footer/index.js
+++ b/pkg/webui/components/footer/index.js
@@ -37,8 +37,8 @@ const Footer = ({ className, documentationLink, links, supportLink, onlineStatus
       <span className={style.claim}>
         <Message content={m.footer} /> –{' '}
       </span>
-      <Link.Anchor secondary className={style.link} href="https://www.thethingsnetwork.org">
-        The Things Network
+      <Link.Anchor secondary className={style.link} href="https://www.thethingsindustries.com">
+        The Things Stack
       </Link.Anchor>
     </div>
     <div className={style.right}>
```

#### How do you propose to test this?

By clicking on the link.

#### Can you do this yourself and submit a Pull Request?

Sure (maybe this should have been a pull request)

kschiffer, 2021-02-15:
Thanks for filing this issue @mattrude!

You are right, this should be `thethingsindustries.com` rather. (cc @johanstokking please confirm)

johanstokking, 2021-02-15:
For TTS open source, I think https://www.thethingsindustries.com/docs makes most sense

PR #3811 (2021-02-16): Update footer link
#### Summary

Closes #3798

#### Changes

- updated footer link from https://www.thethingsnetwork.org/ to https://www.thethingsindustries.com/docs

#### Testing

Manually tested

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Documentation: Relevant documentation is added or updated.
- [ ] Changelog: Significant features, behav [... cut]
closed 2021-02-16 by asmulko