TheThingsNetwork/lorawan-stack #3806: Add LR-FHSS support

Oliv4945, 2021-02-15:
#### Summary

RP2-1.0.2 LoRaWAN® Regional Parameters added LR-FHSS datarates, it would be great to be able to use them with The Things Stack.

#### Can you do this yourself and submit a Pull Request?

No :(

johanstokking, 2021-04-20:
These are the things that need to be done:

- [ ] New modulation `LRFHSS`: https://github.com/TheThingsNetwork/lorawan-stack/blob/b9e3712f32b748b31718896b3e83e153cdc55c6f/api/lorawan.proto#L242-L260. @rvolosatovs 
- [ ] Add LR-FHSS data rates to EU 863-870 MHz, US 902-928 MHz and AU 915-928 MHz, see RP2 1.0.2. Needs #3150. Note that LR-FHSS is only uplink. @rvolosatovs 
- [ ] Support LR-FHSS metadata from gateways. This is only supported for UDP gateways for the moment, as things are being finalized to be integrated in Basic Station from what I understand. Attached is the protocol definition. @KrishnaIyer  [1]

---

[1] Fields added:

```
·         “datr”: The datarate is described as the combination of the Modulation Type, and the Operating Channel Width in kHz.

It is formatted as a string like: “M0CW137”

With:

§  M0: Modulation Type 0 as described in LR-FHSS specification (GMSK, 488Hz)

§  CW137: The OCW as described in the Regional parameters, in kHz

·         “hpw”: A new field for LR-FHSS only, which is the number of hopping steps (grid steps), as described in LR-FHSS specification

It is formatted as a number taking values 52 or 8 depending on the grid index: 0 and 1 respectively.

·         “modu”: The modulation descriptor is set to “LR-FHSS”

·         “codr”: The coding rate, as a string “1/3, 2/3, 5/6…”

·         “freq”: The center frequency of the LR-FHSS channel, as described in the packet forwarder configuration file.

o   A LR-FHSS channel will be described to the gateway as, for example:

"chan_0": { "freq": 867625000, "ocw": "137000", "obw": "488" },

"chan_1": { "freq": 868125000, "ocw": "336000", "obw": "488" },

"chan_2": { "freq": 868600000, "ocw": "137000", "obw": "488" }

o   If a LR-FHSS uplink is received on chan_0, the “freq” field will contain 867625000 in the above example.

o   Note: if no channel defined, or no matching channel has been found it will contain the actual center frequency of the LR-FHSS packet used by the end-device, in MHz.

·         “rsig[x].foff”: the frequency offset compared to the channel frequency. It contains both the “device offset” and the frequency error (xtal, Doppler…). The device offset is unknown by the gateway.

·         “rsig[x].chan”: the LR-FHSS channel the packet has been matched with. (empty if no channel defined in the gateway).

·         “rsig[x].fdri”: The frequency drift on the whole packet, in Hz.

·         “rsig[x].rssic”: RSSI of the packet.

·         “rsig[x].lsnr”: TO BE DEFINED. Not implemented yet.
```

[packet_forwarder_PROTOCOL.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/6342003/packet_forwarder_PROTOCOL.txt)

rvolosatovs, 2021-05-19:
[Proposed implementation](https://github.com/TheThingsNetwork/lorawan-stack/pull/4193)

KrishnaIyer, 2021-07-15:
We have a PR so this is unblocked.

KrishnaIyer, 2021-07-20:
GS part is done via #4413

closed 2022-02-08 by adriansmares