TheThingsNetwork/lorawan-stack #3807: Console Device Overview lacks Array formatting for session keys

lnlp, 2021-02-15:
#### Summary
`Console | <Application> | End devices | <ABP Device> | Overview` lacks array formatting for session keys NwkSKey, SNwkSIntKey, NwkSEncKey and AppSkey.

This is reported as bug because in V2 Console formatters for NwkSKey and AppSKey are available.

#### Steps to Reproduce
See `Console | <Application> | End devices | <ABP Device> | Overview`.
Select a device that uses ABP activation.

#### What do you see now?
The following will show up:

[image]

Select the `Toggle visibility` icon of the different session keys to show their value.
Notice that this will not reveal any `Toggle array formatting` buttons:

[image]

When selecting a device that uses OTAA instead of ABP activation the following will be shown:

[image]

When selecting the `Toggle visibility` icon for AppKey this not only makes the key visible but also reveals a `Toggle array formatting` icon:

[image]

#### What do you want to see instead?
When the keys are made visible by selecting the `Toggle visibility` icon this should also reveal a `Toggle array formatting` icon for each key with corresponding functionality.

#### Environment
Not environment-specific.

#### How do you propose to implement this?
Just add it.

#### Can you do this yourself and submit a Pull Request?
No

kschiffer, 2021-02-15:
This has been fixed by #3752 already, or what do I miss here?

It will be in the next release.

bafonins, 2021-02-15:
Indeed, it is already implemented! Closing this.

closed 2021-02-15 by bafonins

lnlp, 2021-02-16:
I have searched for a similar issue but could not find it.
Apparently #3752 has been fixed and closed 7 days ago already.

If issues are still present in the current product then one is not going to search for existing issues that have already been closed.
From user perspective it will be more useful if issues are closed when the release in which they are fixed has been rolled-out to production.
Or at least be able to see an overview of issues that have been fixed already but are not yet in production.

Closing issues that are not yet in production hides them from users that want to report issues.

kschiffer, 2021-02-16:
Hi @lnlp, I understand your issue but we have to stay pragmatic on the development side. For us the issue is resolved as soon as the fix has been merged. Keeping them open until the release will lead to many issues being kept open well after a fix was released, since it would be require us to go over all the issues once more after the release and we'd likely forget doing so for some.

We are currently working on increasing our release cycles so that the gap between closed issue and release should become smaller in the future.

> or at least be able to see an overview of issues that have been fixed and closed already but are not yet released / not yet in production.

You can do this using the [CHANGELOG.md](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.11/CHANGELOG.md), which also contains the currently unreleased changes at the very top (set to be in the next release).