TheThingsNetwork/lorawan-stack #3808: Go 1.16 "go vet" Issues

htdvisser, 2021-02-15:
#### Summary

...

#### Steps to Reproduce

```bash
$ go vet ./...
```

#### What do you see now?

```
# go.thethings.network/lorawan-stack/v3/pkg/webui
pkg/webui/template.go:50:2: struct field tag `name:-` not compatible with reflect.StructTag.Get: bad syntax for struct tag value
```

```
# go.thethings.network/lorawan-stack/v3/pkg/applicationserver/io/grpc_test
pkg/applicationserver/io/grpc/grpc_test.go:116:6: call to (*T).Fatalf from a non-test goroutine
pkg/applicationserver/io/grpc/grpc_test.go:119:6: call to (*T).FailNow from a non-test goroutine
```

```
# go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io/grpc_test
pkg/gatewayserver/io/grpc/grpc_test.go:132:6: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/io/grpc/grpc_test.go:135:6: call to (*T).FailNow from a non-test goroutine
pkg/gatewayserver/io/grpc/grpc_test.go:198:5: call to (*T).Fatalf from a non-test goroutine
```

```
# go.thethings.network/lorawan-stack/v3/pkg/gatewayserver_test
pkg/gatewayserver/gatewayserver_test.go:791:10: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/gatewayserver_test.go:880:9: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/gatewayserver_test.go:936:6: call to (*T).Fatalf from a non-test goroutine
```

```
# go.thethings.network/lorawan-stack/v3/pkg/errors_test
pkg/errors/errors_test.go:97:7: the cancel function returned by context.WithDeadline should be called, not discarded, to avoid a context leak
```

```
# go.thethings.network/lorawan-stack/v3/pkg/gatewayserver/io/ws_test
pkg/gatewayserver/io/ws/ws_test.go:312:6: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/io/ws/ws_test.go:370:6: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/io/ws/ws_test.go:415:7: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/io/ws/ws_test.go:676:6: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/io/ws/ws_test.go:1013:7: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/io/ws/ws_test.go:1330:7: call to (*T).Fatalf from a non-test goroutine
```

Note: probably not all of these are related to Go 1.16.

#### What do you want to see instead?

No issues

#### Environment

Current v3.11 branch (basically v3.11.0)

Go 1.16

neoaggelos, 2021-02-15:
Assigning myself for

```
# go.thethings.network/lorawan-stack/v3/pkg/gatewayserver_test
pkg/gatewayserver/gatewayserver_test.go:791:10: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/gatewayserver_test.go:880:9: call to (*T).Fatalf from a non-test goroutine
pkg/gatewayserver/gatewayserver_test.go:936:6: call to (*T).Fatalf from a non-test goroutine|
```

rvolosatovs, 2021-02-16:
Don't see anything in NS and related - removing assignment.

closed 2021-03-29 by adriansmares