TheThingsNetwork/lorawan-stack #3810: Display action buttons for inputs next to the input with full text label

kschiffer, 2021-02-15:
#### Summary
Following #2813 and #3522, we will retire the current layout for action buttons of text/byte inputs. Instead we will display action buttons outside the input next to the input.

#### Why do we need this?
- Remove ambiguity for buttons that use icons only
- Unnecessary to restrict ourselves in terms of horizontal spacing, after #3529
- (Action buttons are a bit annoying to handle with input widths)

#### What is already there? What do you see now?
- Current setup that renders the button as part of the input
- Wireframes inside #3522 

#### What is missing? What do you want to see?
Adjusting the markup and styling so that action buttons are displayed outside of the input.

#### How do you propose to implement this?
Trivial. One thing that I think we can improve while at it is the way we currently pass the action to the input:
```
        {hasAction && (
          <div className={style.actions}>
            <Button className={style.button} disabled={disabled} {...action} />
          </div>
        )}
```
This is a bit wonky. I think a better solution is to pass the whole `<Button />` component instead and modify it via `React.cloneElement()`.

#### How do you propose to test this?
I believe we don't use such buttons in our test-cases yet. We should make sure that input actions are tested as well.

#### Can you do this yourself and submit a Pull Request?
This one should be best for @bafonins. This can be part of the ongoing work on end device registration.

lnlp, 2021-02-16:
I assume that actionbuttons for LoRaWAN key fields like 'Toggle array formatting' will be visible but disabled when a key is invisible and will get enabled after key visibility is enabled with a 'Toggle visibility' button?

kschiffer, 2021-02-16:
The component you are referring to are not inputs, but only to display values (we call them `SafeInspector`). These fields will not be changed, at least not with regards to this issue.

If you experience a usability issue with those then please open a separate issue.

kschiffer, 2021-08-02:
Resolved via #4433

closed 2021-08-02 by kschiffer