TheThingsNetwork/lorawan-stack #3813: Improve recognizability and ability to differentiate events and devices by using colors in Live data views

lnlp, 2021-02-16:
#### Summary
Improve recognizability and ability to differentiate events and devices by using colors in Live data views.

It is important that one can recognize and differentiate different event types and which events belong to which devices at a glance. The current bulk of only grey and black text is not very helpful in that aspect.

Proper use of colors can improve this dramastically.
_(With proper use of colors I mean a well designed color scheme that 'does not hurt the eyes'.)_

#### Why do we need this?
To make it much easier to recognize and differentiate between different event types and which events belong to which devices.

#### What is already there? What do you see now?
Only grey and black text.

#### What is missing? What do you want to see?
1. Use different colors for different event types for text displayed in the 'Type' column.
2. In column 'Data preview' use a different color for different device addresses for the DeviceAddr field (and give identical device addresses the same color).
3. Also see #3812

#### Environment
Console. Not environment specific.

#### How do you propose to implement this?
See descriptions above.

#### How do you propose to test this?
\-

#### Can you do this yourself and submit a Pull Request?
No

kschiffer, 2021-02-16:
Blocked on #3817

KrishnaIyer, 2024-10-14:
Let's cover both icons and colors in https://github.com/TheThingsNetwork/lorawan-stack/issues/3812

closed 2024-10-14 by KrishnaIyer