TheThingsNetwork/lorawan-stack #3817: Console event UX framework

kschiffer, 2021-02-16:
#### Summary
We need to review the current UX of event streams in the Console. This issue serves as an overview for me to track individual event related issues and requirements. The goal is to develop a couple of wireframes with a new UX framework under which we can resolve the individual issues below.

#### Why do we need this?
There are currently some UX, as well as technical issues in our event streams:
- [ ] Event streams unable to handle high event frequencies #2887 
- [ ] Event previews are not optimal
  - [ ] Hard to scan for common fields #3716
  - [ ] Horizontal scrolling not optimal
- [ ] "Scannability" of events not optimal (no color-coding, icons not optimal) #3813, #3812
- [ ] Grouping of events and visual indication of related events #3717, #3312
- [ ] Missing filtering of events #2231 
- [ ] Allow proper keyboard navigation for event view #3719
- [ ] Showing advanced and interactive event details #3311
- [ ] Extracting event data as JSON #2383

#### What is already there? What do you see now?
- Many requirements, UX and technical issues
- Some preliminary ideas for solutions

#### What is missing? What do you want to see?
A proper UX framework for events in the Console, defined by a set of descriptive wireframes

#### How do you propose to implement this?
After such framework is found we should continue discussion about implementation details and plan the execution

#### How do you propose to test this?
Testing event related functionality is another big task which should however be discussed separately.

#### Can you do this yourself and submit a Pull Request?
I'll follow up with the UX first and then we can look what other assignments make sense.

kschiffer, 2021-02-16:
I've updated some of the issues above with the blocked label. Not every event related should be blocked on this issue but my rationale is to rather invest into a wider UX concept to solve many issues at once rather than patchworking individual UX issues.

htdvisser, 2021-08-03:
Roadmap update: Q4 2021 - Conference 2022

KrishnaIyer, 2023-07-04:
Blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/6338. 
Let's first discuss if we still want to use the SSE or use WebSockets for handling events from the Console before coming back to this.

KrishnaIyer, 2024-10-14:
Most items here are done, moved or will not be done (commented in the individual issues), Replaced by https://github.com/TheThingsNetwork/lorawan-stack/issues/7341

closed 2024-10-14 by KrishnaIyer